waveform80/picamera #274: pycamera  _get_timestamp  broken?

Berryfier, 2016-04-12:
It may be me just not knowing properly how to, but with picamera.PiCamera as camera I get an error from
   timestamp = camera._get_timestamp()
AttributeError: 'PiCamera' object has no attribute '_get_timestamp'

By extracting the def -get_timestamp() code I can successfully call _get_timestamp to give an answer.

Berryfier, 2016-04-17:
On a Jessie new install, after update I looked at the camera.py in /usr/share/pyshared/picamera/ and see it differs from the camera.py on this GitHub. It seems the attribute '_get_timestamp() is missing from both Wheezy and Jessie raspbian distributions.

waveform80, 2016-04-17:
Not broken; just unreleased as of now. This repo is the primary development repo for picamera - the [current release](https://github.com/waveform80/picamera/releases) is 1.10 and the `timestamp` attribute has been added for the pending 1.11 release (which I'll freely admit is massively overdue - but then that's the price of having kids!).

I'm hoping to get 1.11 out next week as the final piece (a spare CSI adapter for a compute module) has fallen into my lap (thanks to LesP!), but it very much depends on my spare time which is extremely limited at the moment.

Incidentally, once released - don't call `_get_timestamp()` - just query the `timestamp` attribute (the former is the handler for the latter which is a computed property, similar to `resolution`, `framerate`, etc.)

Berryfier, 2016-04-19:
Thank you for your comprehensive explanation. I look forward to the 1.11 release. Your efforts here and in the recipes are of immense help. And I'll feel better knowing I'm making proper use of (at least part of) it!

closed 2016-04-19 by Berryfier