waveform80/picamera #277: Can we do pause ?

CamAuPoint, 2016-04-24:
Hi,
Can we do PAUSE to the camera instead of stop_recording ? So we can PAUSE rapidely ?

Thanks

waveform80, 2016-05-12:
I wouldn't say stop_recording is exactly slow. Testing on my Pi3 (admittedly a reasonably quick platform), stop usually terminates between 1 and 2 frames after the call. How much faster do you want it? (bearing mind you can't go below 1 frame without active truncation...)

lurch, 2016-06-07:
Maybe @CamAuPoint just wants to pause / freeze the preview window, and it's actually the 'resume' (i.e. recalibration of AWB etc.) that they want to happen rapidly? \*shrug\*

waveform80, 2016-06-07:
Hmm ... I've no idea how one might "pause" the preview - disabling the preview connection might do it (but might equally lead to black output). Pausing the actual recording is pretty trivial but resuming will always be relatively slow. There's basically two ways to pause:

1. Stop recording. This is the fastest method, but then resuming the recording means setting up the encoder all over again.

2. Split the recording to something like `/dev/null`. This is much slower because picamera will only split at a key-frame (though in 1.11 it does now request a key-frame from the encoder explicitly). This would be faster to resume, but again requires a key-frame to do so and even with the explicit request that can still take several frames. I doubt it'd be a great deal faster than just starting a new recording (encoder setup isn't exactly "slow").

In other words, I don't know about the preview side of things yet but the output side is already possible and about as fast as I can reasonably make it; I don't see much scope for improvement there.

6by9, 2016-06-07:
Disabling preview connection will flush all buffers, so yes, black screen.

From a GPU level, the expectation would be that client would clear the MMAL_PARAMETER_CAPTURE parameter on output[1], and it will stop outputting buffers. I think that doesn't work with how PiCamera puts together the super-pipeline with multiple encoders.

If it comes off video_splitter then you could use MMAL_PARAMETER_BUFFER_FLAG_FILTER as that sets up for each output port a bitmask of bits that must be present on the buffer for the buffer to be passed on that output. Default is 0xFFFFFFFF (ie all), but can be set to any combination of MMAL_BUFFER_HEADER_FLAG_xxx values. MMAL_BUFFER_HEADER_FLAG_FRAME_END is a rather useful one.

Originally everything was via OpenMaxIL too, and there you could pause the clock component to make sources stop producing data, and unpausing the clock would resume at the same timestamp. Sadly that bit isn't available via MMAL.

lurch, 2016-06-07:
Afraid I dunno how the guts of PiCamera and MMAL etc. work, but could you 'fake' pausing the preview window by taking an image-capture, stopping the preview, and displaying the captured image in the same location as the preview window? And then when 'resuming' stop displaying the captured image and turn the live-preview back on again?

Although that's probably an awful lot of work, so perhaps @CamAuPoint could clarify exactly what it is they want, and why they want it? ;-)  (i.e. what particular application / usage they have in mind)

6by9, 2016-06-07:
If you take a capture and enable burst mode, then the camera component will display the snapshot image created from the capture frame. It will then wait for either burst mode to be cancelled, or a new capture to be requested. Turning that into a generic API is going to be ugly.

waveform80, 2016-06-08:
@lurch on the suggestion of capturing and displaying the capture: yes, it's possible but there's all sorts of complications surrounding that. Firstly it assumes there's a free splitter port for a video frame capture (there might not be). Secondly we'd either need to disconnect the current preview renderer, reconfigure its input port and feed it the data, then when resuming reconfigure it again and reconnect it back to the camera (more delays), or set up a new renderer to display the capture (and there's a limit to the number of simultaneous renderers you can have which is non-trivial to calculate).

I'm afraid this is all sounding "too difficult". Given that the major functionality of pause (i.e. not recording for some period) is already implementable a couple of different ways (and I can't make that much better), and that the "display a frozen image" function is implementable any number of ways in interface layers (Qt, Kivy, pygame, whatever), but that's out of scope for picamera (I'm not going to introduce a dependency on any UI libraries) I'm afraid I have to reject this for now.

waveform80, 2017-02-02:
Closing as I think this one's been satisfactorily answered; do feel free to re-open if you've got further queries about this!

closed 2017-02-02 by waveform80