waveform80/picamera #278: Anyone know if this will work with Pi Camera v2?

anfractuosity, 2016-04-25:
Just wondering if this will work with the latest version of the Pi camera?

cheers
Chris

6by9, 2016-04-26:
https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=145815#p960956

anfractuosity, 2016-04-26:
Thanks @6by9 I'll get one of the new cameras to play with then :)

partach, 2016-04-28:
I have just tested the new camera with picamera v1.10.
Works (after update of rasbian and firmware) ok!
What sticks out:
- image is horizontally flipped with mine and i used to need to do a 180 rotation, now 180 rotation not needed but hflip=True is needed
- think the modes (resolutions) in 1.10 do not support V2 camera yet...

Is there an update of picamera expected that will support V2 functionality of the camera?

6by9, 2016-04-28:
HFlip is a firmware issue that has been fixed. sudo rpi-update.
It was actually missing a VFLIP - H+V = rotate 180.

Likewise the modes currently don't match up - read https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=145815 for the current mode list.
There will be a firmware update hopefully within the next week which will align them slightly more closely (obviously 8MP instead of 5, and the exact frame rates/resolution combinations will vary slightly).

waveform80, 2016-04-29:
@partach yup, working on the update at the moment - the Pi foundation were kind enough to furnish me with a couple of v2 cameras a couple of weeks back which I've been playing with. 1.11 will support the new resolutions, but I'd like to wait for the sensor mode updates (if only so I can get them in the docs prior to the release).

partach, 2016-05-01:
@waveform80 great news and great work :)
I have now installed 3 new cameras (replacing the v1.3) for my home security system with the RPIs. Did a latest update today of the RPI firmware.
(@6by9: that indeed solved the problem for the single one that was flipped and rotated)

Current things that seem not working for me that well yet for the last couple of days:
- Exposure_mode 'night' does not work well on V2 (does not seem to work at all really). With the v1.3 there was a very noticable difference.
- When exposure_mode is set to auto (in the morning, see note below) the camera gets very over exposed  when the sun rises  (almost white screen, nothing visible). It takes some time for the camera (few hours) to recover automatically.
- As with v1.3 i take a snapshot each second with the camera that i can see on an ipad with a regular camera view app. Every 10-30 seconds one of the pics is garbled in different ways each time. I wonder if that is a camera thing or a file system thing (reading the jpg file when it is not fully written?). Not sure if this is related to camera at all but just would like your expertise on this :)

Note:  i have py program that also looks at sunrise and sunsite time and from that sets the exposure_mode twice a day, from auto to night and back.
I did this becuase auto did not work that wel with v1.3 in that it would not automatically go all the way to how far night mode goes.

Hope this helps you too with fixing the open items for V2 :)

Thanks for all your great support guys!

julesrenaud, 2016-05-02:
I don't know if it's normal behaviour (haven't tested with a v1.3), but when recording to a circular stream (default bitrate), the 20 seconds parameter (I'm using the 'Recording to a circular stream' basic recipe) turns into more than a minute of footage before the fake motion event. I've set the resolution at 1280x720 and the resulting file weights about 31MB.

waveform80, 2016-05-03:
@julesrenaud - er, weird place to ask but as it happens: yes, this is normal (read the note at the bottom of the recipe: there's *at least* 20 seconds of data in the buffer - it's a lower bound calculated from the expected bitrate which may be a massive underestimate if the scene contains little motion or complexity). If you require to copy only the last 20 seconds of the content of the buffer you need to look at the frame timestamps (or wait for 1.11 which includes a new [PiCameraCircularIO.copy_to method](http://picamera.readthedocs.io/en/latest/api_streams.html#picamera.PiCameraCircularIO.copy_to) which accepts a seconds parameter)

julesrenaud, 2016-05-03:
@waveform80 Thank you for the response despite my being off topic. I'll wait for 1.11 to try out the copy_to method.

lurch, 2016-06-07:
Perhaps the sections referring to the camera LED in http://picamera.readthedocs.io/en/latest/recipes1.html#controlling-the-led and http://picamera.readthedocs.io/en/latest/recipes2.html#using-a-flash-with-the-camera need a small note regarding the lack of LED on the V2 cameras?

waveform80, 2016-06-07:
Yes - I also need to add something about LED control on the Pi3 (i.e. that I can't yet :)

6by9, 2016-06-07:
Pi3 LED control is just being discussed on https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=109137&start=100#p989832, although that's really a comment for #265 (this issue is good at going off topic!)

waveform80, 2016-06-18:
Closing as this definitely works on the V2 module (the vast majority of the test suite passes - there's a couple of obscure failures, and max. resolution does require gpu_mem increases but that's now added to the docs). It's good enough for 1.11 anyway.

closed 2016-06-18 by waveform80