waveform80/picamera #281: Bug: motion_output do not work with Class/Object which return 0 , is ready solution .

Pe3ucTop, 2016-04-26:
Hello, 

I have noticed that motion_output do not work for object/classes which return 0 . 
Example is empty Deque class with applied write method. 

Inspect encoders.py code and find bug in one place:
Row 836, buggy check:  "if motion_output:" which will return False for empty/0 class.
Other place in code there is correct checking: "if motion_output is not None:" which will return True for empty class.

Solution is to change row 836 of encoders.py: 
from : " if motion_output: "
to: " if motion_output is not None: "

I tried to do update myself/ push commit to git , but it's look like not allowed.

Pe3ucTop, 2016-04-26:
Forgot to mention:
This solution work for me / tested.

waveform80, 2016-05-06:
Good point - I'll get this fixed for 1.11

commit d55c58bd (2016-05-07): Fix #281
closed 2016-05-07 by waveform80