waveform80/picamera #282: HDR support for New Raspberry Pi Camera Board

egoist-sx, 2016-04-29:
Hi, I have noticed that new pi camera module uses sony IMX219PQ, which support ISP HDR.

I saw MMAL_PARAMETER_HIGH_DYNAMIC_RANGE in mmal parameters, but found nowhere used this variable. Is there anyway that I can utilize this new HDR functionality in pi camera?

waveform80, 2016-05-02:
As noted in raspberrypi/userland#311 HDR isn't supported by the new camera. As for what appears in the MMAL header translation - there's plenty of functionality in MMAL which isn't baked into the Pi's version of the firmware (presumably for cost / licensing reasons) which I would assume includes the HDR stuff.

6by9, 2016-05-02:
Correct. HDR is not supported on either of the Pi cameras and it's extremely unlikely to ever be done.
Recommend closing this issue as "won't fix".

closed 2016-05-02 by waveform80