waveform80/picamera #283: Image darker at ISO 200 than ISO 100

Austriker, 2016-05-01:
Hi,

I have been testing Picamera with a fixed shutter_speed. And taking 7 pictures with the 7 ISO.
But the picture at ISO 200 is darker than the image at ISO 100. Is it normal ? Did I missed something ?

I use video_port=True to try to remove the cyan vignetting, that appears in Full Frame.

Here is the code that I use : 
```python
#!/usr/bin/env python3
import picamera
import time

camera = picamera.PiCamera()

camera.exposure_mode = 'auto'

camera.resolution = (2592, 1944)
camera.framerate = 5

iso_list = [100, 200, 320, 400, 500, 640, 800]

camera.shutter_speed = 1185

for i in range(0, len(iso_list)):

    camera.iso = iso_list[i]

    camera.capture(
        img_name,
        use_video_port=True,
        format='jpeg',
        quality=100,
    )

    print("Image {0} taken !".format(img_name))
    time.sleep(1)
```

Thanks

6by9, 2016-05-02:
Which revision of camera board is this with?
Lens shading (cyan vignetting) should be fixed on the V2 sensors, with an option to try the same fix on V1 (add ov5647_acls=1 to boot/config.txt). Or you can use ```still_stats=true``` to also avoid it.
Please use one option or other to other to use the stills port, as that will insert a load more debug into the JPEG, and then post the JPEGs somewhere useful.

Alternatively retrieve the ```analog_gain```, ```digital_gain```, and ```exposure_speed``` for each of your captures.

The only values actually supported are 100, 200, 400, and 800. All others will round to the nearest of the supported list.

Austriker, 2016-05-03:
The camera board is a v1.3
I used a clean raspbian jessie image `2015-09-24-raspbian-jessie` then I did : 

```sh
sudo apt-get update
sudo apt-get upgrade
sudo apt-get dist-upgrade
```
But I didn't do a `sudo rpi-update`

Here is the script I used for the tests : 
```python
#!/usr/bin/env python3
import picamera
import time
import datetime
import os

camera = picamera.PiCamera()

camera.exposure_mode = 'auto'

camera.resolution = (2592, 1944)
camera.framerate = 5

# camera.still_stats = True # for the second test

iso_list = [100, 200, 400, 800]

camera.shutter_speed = 5000

folder = '/home/pi/' + datetime.datetime.strftime(
    datetime.datetime.now(), '%Y-%m-%d_%H-%M-%S_SS15'
)
os.makedirs(folder, exist_ok=True)

for i in range(0, len(iso_list)):

    camera.iso = iso_list[i]

    img_name = os.path.join(folder, '{0}_{1}_{2}.jpeg'.format(
            datetime.datetime.strftime(
                datetime.datetime.now(), '%Y-%m-%d_%H-%M-%S'
            ),
            camera.iso,
            camera.shutter_speed
        )
    )

    camera.capture(
        img_name,
        format='jpeg',
        quality=100,
    )

    print("Image {0} {1} {2} {3} taken !".format(
        img_name,
        camera.analog_gain,
        camera.digital_gain,
        camera.exposure_speed
    ))
    time.sleep(1)
```

The first test is with the `use_video_port=true` : 
[test folder 1](https://drive.google.com/open?id=0BwAaZ8B5G9FPTTRHVUgwdS1mNDg)
```
Image /home/pi/2016-05-03_14-50-18_SS15/2016-05-03_14-50-18_100_1154.jpeg 1 1365/256 1154 taken !
Image /home/pi/2016-05-03_14-50-18_SS15/2016-05-03_14-50-20_200_1154.jpeg 19/16 411/256 1154 taken !
Image /home/pi/2016-05-03_14-50-18_SS15/2016-05-03_14-50-21_400_1154.jpeg 33/16 455/256 1154 taken !
Image /home/pi/2016-05-03_14-50-18_SS15/2016-05-03_14-50-22_800_1154.jpeg 33/8 109/64 1154 taken !
```

The second test with `camera.still_stats = True` and i removed this options `use_video_port=true`:
[test folder 2](https://drive.google.com/folderview?id=0BwAaZ8B5G9FPTXBqbVRKdFJETmc&usp=sharing) 
```
Image /home/pi/2016-05-03_14-52-05_SS15/2016-05-03_14-52-05_100_1154.jpeg 1 281/256 1154 taken !
Image /home/pi/2016-05-03_14-52-05_SS15/2016-05-03_14-52-08_200_1154.jpeg 19/16 257/256 1154 taken !
Image /home/pi/2016-05-03_14-52-05_SS15/2016-05-03_14-52-10_400_1154.jpeg 15/8 71/64 1154 taken !
Image /home/pi/2016-05-03_14-52-05_SS15/2016-05-03_14-52-12_800_1154.jpeg 31/8 275/256 1154 taken !
```

I did the same test with the option `ov5647_acls=1` but it doesn't remove the vigneting.
The first test shoes the defect at ISO 200. But with the second test the pictures works properly and the vigneting seems to disapear.

Delgan, 2016-06-21:
I made some tests with the new camera V2.1 and I can confirm @6by9 words, whatever the values of `still_stats` and `still_stats` was, I did not notice any cyan vignetting.

Moreover, the iso parameter was acting as expected and I did not encounter weird behavior: the image brightness looked to enhance while iso was increasing (except maybe for iso 100/200 with `use_video_port=true` which look the same).

Images resulting from this test can be see here:
* [`camera.still_stats=False, use_video_port=False`](https://drive.google.com/folderview?id=0B7LX4AhaYGhNOE1IR1Flb2ZZRzA&usp=sharing)
* [`camera.still_stats=True, use_video_port=False`](https://drive.google.com/folderview?id=0B7LX4AhaYGhNajNieWo1ak9mSXc&usp=sharing)
* [`camera.still_stats=False, use_video_port=True`](https://drive.google.com/folderview?id=0B7LX4AhaYGhNZXh1OEtUVHcyUXc&usp=sharing)
* [`camera.still_stats=True, use_video_port=True`](https://drive.google.com/folderview?id=0B7LX4AhaYGhNck1KYUdwdGJvWHM&usp=sharing)