waveform80/picamera #284: PiCameraRuntimeError: Timed out waiting for capture to end with V2 camera

partach, 2016-05-13:
Hi 

When setting up V2 camera I increased the resolution in my capture SW.
This led (either the new camera or higher resolution) to great instability. It only runs about 5-10 hours before my every second capture freezes and is not update anymore.
The log file is shown below.
I searched a bit online and found two tips:
- increase a buffer somewhere (dont know how)
- set time out to 60 seconds (as shown in log file that still gives issues)

Strange thing for me is it runs perfectly for 5-10 hours and then gets into non-recoverable trouble.
Some memory leak somewhere?
Any ideas from you are welcome to help solve this :)
Thanks in advance!

Log:
2016/05/11 14:32:31 Initialise picamera
2016/05/11 14:32:32 starting main loop picamera
2016/05/11 19:47:15 Error: Traceback (most recent call last):
  File "/home/pi/mypicamera.py", line 199, in webcamCaptures
    camera.capture_sequence([snapf, pictureFN], use_video_port=False, burst = True, splitter_port=3,quality=6,resize=(video_width,video_height))
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 1682, in capture_sequence
    'Timed out waiting for capture to end')
PiCameraRuntimeError: Timed out waiting for capture to end

2016/05/11 19:48:15 Error: Traceback (most recent call last):
  File "/home/pi/mypicamera.py", line 201, in webcamCaptures
    camera.capture_sequence([snapf], use_video_port=False, burst = True, splitter_port=3,quality=6,resize=(video_width,video_height))
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 1682, in capture_sequence
    'Timed out waiting for capture to end')
PiCameraRuntimeError: Timed out waiting for capture to end

2016/05/11 19:49:15 Error: Traceback (most recent call last):
  File "/home/pi/mypicamera.py", line 199, in webcamCaptures
    camera.capture_sequence([snapf, pictureFN], use_video_port=False, burst = True, splitter_port=3,quality=6,resize=(video_width,video_height))
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 1682, in capture_sequence
    'Timed out waiting for capture to end')
PiCameraRuntimeError: Timed out waiting for capture to end

6by9, 2016-05-13:
Probably not the same, but there is an issue being discussed on the forums where the sensor stops responding after a few hours - https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=147744

I'd expect PiCamera to check the return value from mmal_component_enable and fail earlier than this, but is your use case repeatedly opening and closing the camera, or running with the same instance?

partach, 2016-05-13:
only once i create an instance (camera) of the picamera.

Here a snapshot of that part of the code:

with picamera.PiCamera() as camera:
   saveLogData("Initialise picamera")
   camera.resolution = (video_width, video_height)
   camera.framerate = video_framerate
   camera.rotation = 0
   camera.hflip = True
   camera.vflip = video_vflip
   camera.video_stabilization = video_stabilization
   camera.annotate_background = video_annotate_background
   camera.annotate_frame_num = video_annotate_frame_num
   camera.CAPTURE_TIMEOUT = 60 # seconds
   # this is the video where motion detection is executed on
   camera.start_recording('/dev/null', splitter_port=2, resize=(motion_width,motion_height) ,format='h264', motion_output=DetectMotion(camera, size=(motion_$

   # setup thread to keep auto tracking of daytime for exposure
   daytimechkthread = threading.Thread(target=keepTrackofDayTime, args=[camera])
   daytimechkthread.start()

   # setup webcam feed file
   snapfilename = filepathsnap + "/" + snapshot_capture_filename
   snapf = open(snapfilename,'wb')
   FNT="default"
   saveLogData("starting main loop picamera")

   try:
     while True:
       timeisnow = dt.datetime.now().strftime('%Y%m%d-%H%M%S')
       # latest text to show in camera image
       camera.annotate_text =  timeisnow + " " + movement_mode_annotation + exposure_mode_annotation + " " + "md %d, mq %d" % (motion_detected, motion_quite)
       if getMotionLevel() == 1: # init motion is set
          setMotionLevel(2) # motion in progress
       if getMotionLevel() == 2: # motion in progress
          webcamCaptures(snapf,camera,filepath + "/" + timeisnow + ".jpg")
       elif getMotionLevel() == 3: # motion progress is stopped
          webcamCaptures(snapf,camera,filepath + "/" + timeisnow + ".jpg")
          setMotionLevel(0) # no motion in progress
       webcamCaptures(snapf,camera) #standard picture for security cam feed
   except:
       ErrorHandling()
   finally:
       snapf.close()
       camera.stop_recording(splitter_port=2)
       setMotionLevel(0) # no motion in progress
       saveLogData("we have stopped with picam")

pootle, 2016-07-18:
I get this trying to get the exposure time up to near 10 seconds, it looks like the various things (even in capture continuous ) that precede and follow a capture can exceed the limit built in to the python support. Up to about 8 seems seems to be safe, the it keels over fairly quickly.

6by9 suggested in t[his pi forum thread ](https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=154662)that the limit needs to increase.

waveform80, 2017-02-02:
Looks like CAPTURE_TIMEOUT needs bumping up; I'll get this included for the next release

commit 2c2917d5 (2017-02-14): Fix #284

Increase CAPTURE_TIMEOUT for 10 second captures on the V2 module
closed 2017-02-14 by waveform80