waveform80/picamera #285: any beta testers needed for 1.11?

partach, 2016-05-20:
Offering to help. I have 4 PIs with cameras and 3 new V2 camera's in total

julesrenaud, 2016-05-20:
+1
I have 2 PIs and 2 v2 camera boards

waveform80, 2016-05-21:
If anyone wants to beta-test the stuff coming in 1.11, please feel free to do so - just follow the "development instructions" in the [installation instructions](http://picamera.readthedocs.io/en/latest/install.html) and you'll have a clone of the stuff I'm working on for 1.11. However, be aware there is still quite a bit broken; behind the scenes 6by9 has been beavering away on the firmware fixing things and I'm hoping to take advantage of those fixes in 1.11.

Unfortunately, the stuff getting fixed doesn't "fail nicely" in the current firmwares. To give an example, let's say you've got a video splitter attached to the camera's video port (which picamera always does) and you run that splitter's input in I420 (which picamera 1.10 and below does). Now, you request RGB output from one of the splitter's output ports and trigger a capture. Everything will appear to work normally (no errors, the capture will run) but you'll get I420 (YUV) data out. Same with BGR and anything else you request. And there's a few more things like this getting cleaned up.

Hopefully fixes for these will be in a forthcoming firmware but it'll probably be a while before it gets into Raspbian stable, so at picamera 1.11 release (whenever that is), the situation is probably going to be something like this:

* For the majority of simple use cases (video recording, JPEG image capturing, previews, etc.) things will work just fine with old firmwares so *most* people won't have much pain with 1.11

* Unfortunately, for people doing more esoteric things (RGB captures, motion vector capture, etc.) they will *have* to run `rpi-update` otherwise things will break in interesting and annoying ways (this is also the situation you'll probably find playing with 1.11 at the moment, as bits of it assume as-yet-unreleased firmware stuff).

So - feel free to test, but you'll probably find a fair number of things broken right now (i.e. don't post tickets for that stuff - if it's in the test suite, I'm aware of it - although if it's not in the test suite, I might not be, so that might be worth a ticket :)

waveform80, 2017-02-02:
1.11 got released, so I'll close this but feel free to play with the up-coming 1.13 stuff - things are rapidly nearing completion now!

closed 2017-02-02 by waveform80