waveform80/picamera #286: Preview window aspect ratio is wrong

laser101, 2016-05-24:
Not sure if its just me, but the picamera preview window appears to be squashed vertically. I'm trying to create a square preview window with width=155px, and height=155px, in the top right of my screen.

```
camera.start_preview( fullscreen=False, window=(1093,32,155,155) )
```

However, when this overlay is rendered on to the screen it appears to have the correct width, but the height is quite a lot smaller than 155px. Its difficult to measure, but it looks like the ratio might be 4:3, and my screen size just happens to have ratio 4:3. Could this be a coincidence?

Not sure whether its relevant, but I'm using the zoom functionality to capture a region of interest from the camera with exactly 155px x 155px, hence the small preview window. I was expecting the 155px x 155px region of interest to be rendered square in the overlay preview.

Any thoughts?

6by9, 2016-05-24:
You haven't changed the resolution of the pipeline, you've just set the output images to be rendered to a 155x155 window with the aspect ratio retained.
Set ```camera.resolution = (480,480)``` first to request the pipeline to produce a 1:1 aspect ratio image, and then it should fill your window. (155 won't work on the pipeline as it isn't an even value, so YUV chroma subsampling becomes an issue).

laser101, 2016-05-24:
Thanks for your speedy response, excellent.

I've modified the resolution as suggested, and you are correct, this does fill the preview window as described.

Unfortunately this doesn't quite move me in the right direction. I'm trying to capture from the sensor in mode 2 (full frame), and then extract a ROI (region of interest) according to user supplied coordinates. The mode 2 aspect ration is 4:3, which I guess is what is causing the preview overlay to be rendered with that ratio.

In the test case I posted, I am capturing a square ROI from the fullframe, but the preview overlay is being rendered with 4:3 aspect ratio. Is there anything I can do to force the preview window to have 1:1 aspect ratio without changing the sensor resolution?

As a thought experiment, consider that the user requests a square region of interest (100px x 100px) from the top-left corner of the fullframe (mode 2). I can capture this region using the zoom functionality, but I'm not sure how to render it square in the preview window. Unfortunately don't think I can use a square resolution setting (as you suggested), since I would lose access to the top-left corner of the sensor.

Hoping that I haven't got confused somewhere along the line,

R

6by9, 2016-05-25:
The sensor output is cropped to the requested aspect ratio first, then further cropped according to digital zoom / ROI. So you are right that if you've asked for a 1:1 aspect ratio output and mode 2, then the left and right (2592-1944)/2 = 324 pixels of each line are not available. That's just the way the GPU system is set up.

What may be an option (and I haven't checked it) is to set up a src_rect within the renderer. 
If I've got it right, that allows you to ask for a 4:3 aspect ratio image from the camera, but then tell the renderer to only display part of that image. So camera can produce 640x480, and you then ask for it to display x=0,y=0,w=480,h=480 to pick up a 1:1 region from the left side of that image.
It looks like waveform80 has support in there via http://picamera.readthedocs.io/en/release-1.10/api_renderers.html#picamera.renderers.PiRenderer.crop
Requires a little more fiddling to get correct, but should give you your desired result.

closed 2017-02-02 by waveform80