waveform80/picamera #288: Efficient motion detection (comparing captures vs. using PiMotionAnalysis)

maximedupre, 2016-05-31:
The documentation suggests two ways to implement motion detection:

1. [By comparing captures](http://picamera.readthedocs.io/en/release-1.10/recipes2.html#splitting-to-from-a-circular-stream)
2. [By using PiMotionAnalysis as the motion ouput](http://picamera.readthedocs.io/en/release-1.10/recipes2.html#custom-outputs)

Which motion detection solution is the best/more efficient?

My guess is that using PiMotionAnalysis is a better choice, but since comparing captures is also suggested, there must be some advantages over PiMotionAnalysis?

Thank you for the enormous value that is provided with Picamera and for the support!

waveform80, 2016-05-31:
It's probably about time I wrote something on this! I'll preface this by saying I'm not a motion detection expert (or an imaging expert generally), but here's what I do know about the various methods:

## Motion detection via PiMotionAnalysis

In this case we're getting the camera to record H.264 video. Being a heavily compressed video format, this relies on detecting changes between adjacent frames to enhance compression. In order to detect changes between frames, the encoder (in the GPU) constantly runs a "motion estimation" task. This splits the two frames it's comparing into "macro-blocks" (16x16 pixel blocks), and tries to figure out which blocks moved between the frames, and by how much. Note that the motion estimation is fairly crude:

* It detects X and/or Y translation
* It does *not* detect rotation
* It does *not* detect translation in the Z axis (movement "into" or "out of" the frame).  For  example, movement into the frame would be represented as lots of macro-blocks moving toward the edges of the frame.

The output of the motion estimation task is an array of X, Y, and SAD values for each macro-block. The SAD value is a 16-bit "[Sum of Absolute Differences](https://en.wikipedia.org/wiki/Sum_of_absolute_differences)" which is an estimate of how well the moved macro-block matches its original. The Pi's camera firmware allows access to this data (uniquely, to my knowledge) which means we can use it for our own purposes, like motion detection.

Advantages:

* The GPU is doing the *vast* majority of the work here, so until we get to operate on the motion vector data, the CPU will be essentially idle (beyond picamera's usual overhead).
* The motion estimation is rather good at detect fairly large objects moving in the frame

Disadvantages:

* Things smaller than a macro-block probably won't be noticed
* Only detects X/Y motion "well" - more complex movements like the rotation of a single macro-block probably won't be noticed

One more thing to bear in mind is that the motion estimation is intended for video compression (and for that, it's perfectly good enough). This has some interesting consequences, but here I'll warn that we're moving into the realms of conjecture: I don't *know* how the motion estimation works under the covers but the following is theoretically possible.

Consider two frames containing objects like so:

```
+---------------+   +---------------+
|               |   |               |
|               |   |               |
| XX  XX        |   |     XX  XX    |
|               |   |               |
|               |   |               |
+---------------+   +---------------+
```

There's two ways of representing the motion here:

1. Both blocks were translated with X=4, Y=0
2. The left block was translated X=8, Y=0, while the right block stayed still

Both are perfectly valid representations of the motion. Which one is true? We Just Don't Know! However, it's worth noting that the second representation may be more efficient for the encoder. The only reason I mention this is to caution about using the data for anything more than motion estimation: trying to figure out *what* motion actually occurred may be more or less successful.

## Motion detection via image captures

This one's a lot simpler as it's basically down to you to implement the motion detection in any way you like. Background subtraction, erosion, averaging, throw in whatever you like. Use whatever size macro-blocks you want - go pixel perfect if you think it's really worth it!

The disadvantage is obviously that the vast majority of the work will be done on the CPU and probably in Python (although if you're using numpy / opencv you're still mostly in compiled code, if you're careful), so it'll be commensurately slower.

That in turn may require the use of a more powerful Pi (model 2B, 3B, etc.) while motion estimation will run happily on a Pi Zero if you want (the same GPU is in all Pi's - it's just over-clocked a little on the 3B).

## Conclusion

So, that's the trade-off, basically: speed (motion estimation vectors) vs accuracy and control (capture comparisons). But remember you can run multiple things at once over the splitter, so you may even want to try combining the approaches.

Anyway, I'd be most interested to see what people come up with in this area! I don't get enough time to play with this stuff, and it's always fascinating :)

maximedupre, 2016-05-31:
This clears up a lot of things. I could not have expected a better answer. Thanks!

maximedupre, 2016-06-16:
I have started to learn more about macro-blocks (thanks to the links you put in the documentation) and although it has cleared up some things, there are things that still confuse me. I know you said that you are not a motion detection helper, but perhaps you already know the answer to my questions.

1.  In the `(x, y, sad)` structure, `sad` is the Sum of Absolute Differences, but what are `x` and `y`? Do they mean "translation that occurred on the x/y axis" (I assume that the answer is yes for my following questions)? Are their value in pixels?
2. What does it mean when you have a macro-block that looks like this: `(0, 0, 202)`? Does it mean that the pixels in the macro-block have changed quite a lot, but for some reason there has not been any transition from another macro-block (something just appeared right in the macro-block)?
3. What does it mean when you have a macro-block that looks like this: `(20, 25, 300)`? It seems like the macro-block has been translated and that it is very different from the original macro-block. Can we say that this macro-block is the "closest match" to the original, but that since it has a high `sad`, so is the margin of error and that we should not consider it as a motion?
4. Why are you not using the `sad` value in your [motion detection algorithm](https://picamera.readthedocs.io/en/release-1.10/api_array.html#pimotionanalysis)s?

waveform80, 2016-06-17:
1. My understanding is that it is indeed the x/y axis, and yes the values are definitely in pixels (according to [gsh's post here](https://www.raspberrypi.org/forums/viewtopic.php?p=548816#p548816)).
2. I'm only guessing at this point as I don't know the details of the Coarse Motion Estimation (CME) algorithm but I'd guess that it means the block doesn't match the prior frame (high SAD) but that no matching block could be found within +/- 128 pixels in either axis (the vectors are signed 8-bit values so they never exceed +/- 128 which I therefore also assume is the limit of the motion search).
3. I would guess that means that the closest matching block is 20 pixels across and 25 pixels down, but that it's not a very good match (still for the purposes of video compression it might be good enough - remember that that's the ultimate purpose of the CME).
4. Because they're purely crude examples for people to build upon. I've not had the time to spend building a really good motion detector but if people want to contribute a better example to docs, feel free!

Just bear in mind that the docs are meant to be easily digestible, so I tend to try and keep the examples short and easily understood. There's some more comprehensive examples (which I deemed a little too long for the docs) in the [picamera_demos](https://github.com/waveform80/picamera_demos) repo, but again nothing terribly impressive for motion detection. Pull requests gratefully received!

waveform80, 2017-02-02:
Closing old issues - I think this one's adequately answered but do feel free to re-open if you've further questions!

closed 2017-02-02 by waveform80