waveform80/picamera #290: Picamera V2 & Photo Booth resolution problems

mikeredwards, 2016-06-04:
Dear all,
I have been looking around for a while for a solution to my problem, but can't seem to find any threads that give me the right solution.
I have a picamera hooked up to my raspberry pi which is connected via HDMI to my TV. This and some other hardware form a Photo Booth that I made for my wedding and that I will now lend to my friends and family for their events
The problem is the difference between the preview resolution and the photo taken, and the fact that I would like maximum resolution pictures. 
When I don't specify a resolution, the photos are taken at 1600*900, and there is already a difference between the preview and the photo. The photo is zoomed out.
When I specify a resolution matching the TV (1920*1080), I get the same thing, slightly more zoomed out.
Now that I have the picamera V2, I would like to take advantage of the high resolution, so I specified a resolution with the same ratio as the TV : 3280*1845. The photos are even more zoomed out
I have understood that the preview overlay draws a rectangle from the centre of the screen, and this rectangle is a PoV of the FoV. I have understood that the use_video_port function could be used to match the preview and the photo, but this doesn't give me hi-res pictures. I have looked at the crop/zoom function, but this only allows me to zoom in even more.
Could anyone help me ? The ratios being the same, I would like to get a perfect match between the preview and the capture
Thanks in advance

6by9, 2016-06-04:
Please read http://picamera.readthedocs.io/en/latest/fov.html#camera-modes for details of the various sensor modes, including for the V2 camera.
At a guess you're running old firmware, as the latest firmware includes a 1620x922 mode that should be selected if you're asking for 1600x900 on the display. 1920x1080 has always been cropped on both v1 and v2 sensors, as the hardware can't achieve 30fps.

Basic summary is to keep the preview resolution below 1600x900 if you want to achieve full FOV.

mikeredwards, 2016-06-05:
OK, that's what I sort of imagined. I will think about getting a higher resolution monitor if I want to display a better resolution.
Thanks for your help

blackbicbiro, 2016-06-23:
what is the outcome of this? I have the same problem with the preview not being the same resolution as the picture taken. Is it possible to drop the frame rate of the preview to give the same size as the still pictures that will be taken if its a hardware issue. im trying to take a 2592, 1944 still but my preview is not close to that. I am using 1.3v camera

mikeredwards, 2016-07-06:
The outcome is that you have to find the mode that is closest to the resolution of your screen otherwise there will be a difference. Depending on the resolution of your screen check the link sent by 6by9 to see what mode to apply

closed 2016-07-06 by mikeredwards

6by9, 2016-07-14:
Not really.
Choose a resolution that matches your capture aspect ratio (even if that doesn't match your screen), and is <1280x960 if you want the FOV to match (you need to avoid the 1080P cropped sensor mode).