waveform80/picamera #292: No preview window visible on a minimal installation

yuvadm, 2016-06-05:
When running the most basic example on a clean minimal installation running Arch Linux ARM on a Raspberry Pi 3, the preview window simply doesn't show up at all:

```python
import picamera
with picamera.PiCamera() as camera:
    camera.start_preview()
```

a preview render object is returned, but no preview window is shown on screen. This happens regardless of running in a plain terminal, or from inside an xterm window running on Xorg.

How can I debug this problem? It's clear that a library might be missing, but there's no clue as to which library that actually is.

yuvadm, 2016-06-05:
Interesting update, when running `raspistill -v` from plain shell I actually get a preview render window!

yuvadm, 2016-06-05:
This was a stupid testing mistake. The preview ended right after the entire `with` block executed, so nothing was rendered on screen. After adding a few seconds of sleep it turns out the preview works just fine.

closed 2016-06-05 by yuvadm

waveform80, 2016-06-05:
Ah, no worries - this is one of the reasons I've switched up the [basic recipes chapter](http://picamera.readthedocs.io/en/latest/recipes1.html) in the forthcoming release to avoid using "with" - it's confusing for people either new to python or programming generally and really isn't necessary for basic use-cases. It's still in the advanced recipes chapter, as it is useful in some scenarios and avoids most cases of "creating the camera twice" but I wanted to make it clear that for stuff like this it's absolutely fine to just use "camera = picamera.PiCamera()"

flagtrax, 2017-09-06:
Forgive me but I'm having trouble understanding the concept. Not python but the Rpi camera  function. I just received a camera and started playing with it; using the example on the Raspberry pi dot org website. I couldn't see a preview with their example of:
raspistill -k, so googled and found python examples. those didn't work either. I did however  have a saved file of "test.jpg" created even without a preview. What am I not understanding? Any help greatly appreciated, the Raspberry pi org documentation is more confusing than helpful at least to me.

waveform80, 2017-09-07:
@flagtrax does the "test.jpg" contain a valid image? How are you viewing the Raspberry Pi's output? The preview only appears on devices attached to the HDMI, composite, or DSI ports.

flagtrax, 2017-09-08:
Hello Dave, and thanks so much for the reply. I've done some checking and
found that there was a valid image. The output was to an HDMI monitor. I
also tried at one point to do a "restore" by holding the shift key at boot.
(on the Pixel desktop ver) which caused the OS to basically blow up. I
downloaded the latest NOOBS and installed it, and from that point, I find
that all seems to work as supposed to. (At least as far as I've gotten with
it). I am quite pleased with the camera's performance, and look forward to
experimenting with streaming etc. Again, thanks for the reply.
John

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>