waveform80/picamera #293: PiCamera hangs during initialization if run with multiprocessing

muodov, 2016-06-06:
It seems that `PiCamera` constructor blocks forever if called in subprocess using `multiprocessing` module. Here is a simple proof-of-concept code:

```python
from multiprocessing import Process

def preview():
    with picamera.PiCamera() as cam:
        cam.start_preview()
        time.sleep(3)

p = Process(target=preview)
p.start()
```

The started process will hang during PiCamera initialization. I tried to narrow down the problem and it seems that it never returns from calls to underlying library functions: for example `bcm_host.graphics_get_display_size()` here: https://github.com/waveform80/picamera/blob/master/picamera/camera.py#L375

On the latest development version (https://github.com/waveform80/picamera/commit/9f523b535f4ae0311568337eed1d1eb633d0edf6), specifying `resolution` argument works as a workaround because that piece of code is skipped, and initialization finishes. However, that is not the case for the version available in PyPI: that one hangs somewhere around `mmal*` calls in this function: https://github.com/waveform80/picamera/blob/6f36295e3ecd7543f633d3770bc7ef08840e3611/picamera/camera.py#L509

waveform80, 2016-06-07:
Strange ... I had to add a few imports to your test script to get it working, but it did work quite happily on both picamera 1.10, and the current master for 1.11. This was also on a Pi3, using Python 2.7 and 3.4 (default versions on Raspbian). The firmware was release 884 (which is pretty recent) so you might want to try `rpi-update` to see if that fixes things for you. For reference, here's the test script I ran:

```python
from picamera import PiCamera
from time import sleep
from multiprocessing import Process

def preview():
    with PiCamera() as cam:
        cam.start_preview()
        sleep(3)

p = Process(target=preview)
p.start()
```

waveform80, 2016-06-07:
I should add - it's a bit unusual to want to open a PiCamera instance in a sub-process. The camera can only be opened by a single process at a time so if you wanted to use parallel processing with captured data you'd more likely want to have the "master" process capturing stuff from the camera, and feed the captured data to the sub-processes for work via some IPC mechanism.

muodov, 2016-06-07:
In my case I have a kivy-based GUI that spins up a camera-aware process and uses an RPC-like protocol based on `multiprocessing.Pipe`.
It could be that the combination kivy + multiprocessing + picamera is giving me a deadlock, but I'm pretty sure I could reproduce it yesterday with the snippet above.
Let me just try it again and get back with more details

muodov, 2016-06-08:
I checked again, and indeed, sample code above seems to work... 😳
However, my real kivy app doesn't. I ended up with a separate launcher script that would spin up two child processes: one with kivy GUI, and another with camera code. That seems to solve my problem.

My guess is that kivy graphic initalizations conflict with picamera somewhere, so one doesn't work well with another in the same python context. Maybe Raspberry Pi touchscreen has something to do with it as well, because that's what I'm using.

Anyway, I'll close this for now, since the issue is not clear (it could even be not picamera problem). Perhaps I'll find time someday to make a working PoC.

Sorry for disturbing and thanks for a great library 😉

closed 2016-06-08 by muodov

waveform80, 2016-06-09:
No problem at all - it's worth knowing that there's some incompatibility with kivy. I've had a (very brief) trawl through the kivy source to find stuff that might potentially conflict, and there's some interesting bits in lib/vidcore_lite/ but how exactly they (might) conflict I couldn't say at the moment (while I'm now fairly well versed in MMAL, I know very little about dispmanx which it's playing around with at the bottom of egl.pyx).

Anyway - sorry there's not much I can do about this for now, but if I get the chance to dig into dispmanx in the future I may come back to this!