waveform80/picamera #295: MMAL error when trying to record an H264 in 2592x1944

maximedupre, 2016-06-17:
It might be related to #294, however I think this issue is only for the v2 cameras.

I have a revision 1.3 Pi Camera and I'm using `picamera` version 1.10.

When calling `start_recording` for `camera = picamera.camera.PiCamera(resolution = (2592, 1944))`, I get those 3 errors in my terminal:

    mmal: mmal_vc_port_enable: failed to enable port vc.ril.video_encode:in:0(I420): ENOSPC
    mmal: mmal_port_enable: failed to enable connected port (vc.ril.video_encode:in:0(I420))0x21eba40 (ENOSPC)
    mmal: mmal_connection_enable: output port couldn't be enabled

All the other resolutions listed in the camera modes table work fine.

Also, what's the difference between the camera modes 2 & 3? I can see that the framerate isn't the same, but how do I know which one I'll get?

Cheers!

waveform80, 2016-06-18:
> When calling start_recording for camera = picamera.camera.PiCamera(resolution = (2592, 1944)), I get those 3 errors in my terminal:

H264 recordings are limited to 1080p; I don't recall the precise details but it's something to do with a hardware limit on the frame width. Beyond 1920 pixels wide (or some figure near that), the GPU simply doesn't have the capacity to encode H264 (the GPU was likely designed with a 1080p30 limit for H264 in mind).

> Also, what's the difference between the camera modes 2 & 3? I can see that the framerate isn't the same, but how do I know which one I'll get?

On the V2 camera, there's no difference at all, so up to 15fps you'll presumably get mode 2. On the V1 camera, the mode selected will be dictated by the framerate (change to a framerate <1fps and you'll get mode 3, >1fps you'll get mode 2). Note that resolution and framerate *both* dictate mode. For example, if you set resolution to max and framerate to >15fps you'll get mode 1 (cropped FoV) up to 30fps. Above 30fps you'll probably get mode 4 (full FoV), and at some point above that you'll get mode 6.

maximedupre, 2016-06-18:
> H264 recordings are limited to 1080p; I don't recall the precise details but it's something to do with a hardware limit on the frame width. Beyond 1920 pixels wide (or some figure near that), the [...]

I see. Is it suppose to fail only for H264 recordings? I tried a MJPEG recording and got the same error. If there is no possible way to record a video in this resolution, isn't the [camera modes documentation](https://picamera.readthedocs.io/en/release-1.10/fov.html#camera-modes) confusing?

waveform80, 2016-06-18:
MJPEG can record at higher resolutions that H.264 (less work for the GPU to do presumably) but even it will top out before 2592x1944. It *is* possible to record unencoded video (YUV, RGB, etc.) at that resolution but the bandwidth requirements are sufficiently large that either you need a very low framerate or your processing needs to be minimal (or the output will need to be something with higher bandwidth like Ethernet or a circular buffer in RAM).

The camera modes chapter doesn't list every limitation of the camera partially because I don't know them all (I'm no expert on the firmware or hardware, and that chapter is largely made of tidbits I've gathered together from forum posts or conversations with the firmware devs), and partially because writing digestible documentation sometimes means leaving things out.

That said, I probably ought to add a chapter to the docs about video recording details. It's an area most people find very confusing (partial FoVs in 1080p, ridiculously high bitrate limits, the way quality gets better the lower the value gets, etc. etc.)

Documentation PRs are also most welcome!

waveform80, 2016-06-18:
I should add it's also possible to run the camera at maximum resolution, perform an H264 recording with a resizer in the pipeline for a lower resolution and capture a full resolution JPEG frame from a different splitter port when requested.

maximedupre, 2016-06-18:
Ok perfect! Thanks for the info!

closed 2016-06-18 by maximedupre