waveform80/picamera #297: cannot capturing to an OpenCV object

tsuncp, 2016-06-20:
I am using the latest version 1.11 of picamera , 1.11.0 of numpy and python 3.4

I follow the code from document 

```
import time
import picamera
import numpy as np
import cv2

with picamera.PiCamera() as camera:
    camera.resolution = (320, 240)
    camera.framerate = 24
    time.sleep(2)
    image = np.empty((240, 320, 3), dtype=np.uint8)
    camera.capture(image, 'bgr')
```

But I have got 
TypeError : startswith first arg must be bytes or a tuple of bytes , not str

I also upgraded to the latest firmware.

jrosebr1, 2016-06-24:
Confirmed, I'm also seeing the same error.

waveform80, 2016-07-02:
Finally found some time to look into this - it's strange given that the test suite *should* have picked this up. Probably means there's some issue in my test setup...

waveform80, 2016-07-02:
Damn. Well, that was annoyingly easy. There was indeed an issue in the test setup: the box I thought was running the suite under Python 3 ... was running it under Python 2 (so I was just testing the library twice under Python 2); I'd switched to using two boxes for testing to speed things up a bit (given the test suite takes ages to run) but the 3 box had the old venv from the 2 box as well as its py3 venv, and the test config was still pointing at the old venv.

The actual bug itself is also annoyingly trivial ... here's the patch:
```patch
diff --git a/picamera/encoders.py b/picamera/encoders.py
index 73ff360..c4a69f9 100644
--- a/picamera/encoders.py
+++ b/picamera/encoders.py
@@ -531,7 +531,7 @@ class PiRawMixin(PiEncoder):
         # port format is set (height is aligned too, simply for consistency
         # with old picamera versions). Warn the user as they're not going to
         # get the resolution they expect
-        if not resize and format != 'yuv' and input_port.name.startswith('vc.ril.video_splitter'):
+        if not resize and format != 'yuv' and input_port.name.startswith(b'vc.ril.video_splitter'):
             # Workaround: Expected frame size is rounded to 16x16 when splitter
             # port with no resizer is used and format is not YUV
             fwidth = mmal.VCOS_ALIGN_UP(width, 16)
```

Yup ... that's it. A missing "b". I'll push the fix in a mo, and this is probably serious enough to warrant a point release to fix it, but I'm not sure I've got time for that this weekend.

jrosebr1, 2016-07-02:
Nice, congrats on resolving the issue! For what it's worth, I would definitely agree that this is worth a point release to fix. Based on [this post alone on PyImageSearch](http://www.pyimagesearch.com/2015/03/30/accessing-the-raspberry-pi-camera-with-opencv-and-python/) I get at least 1 comment per and 2-3 emails per day asking about the error. Updating to 1.12 will at least ensure that a standard pip install will work on both Python 2.7 and Python 3.

waveform80, 2016-07-02:
Okay - I'll try and find some time to get another release cranked out tomorrow. Hopefully, little enough will have changed from the last release that the packaging scripts still work (I swear, every time I go to build packages *something* changes to break my stuff - though last time it was PyPI instead of debuild!). Just finished a full run of the test suite under py3 and there's a couple more issues that look similar (bytes vs unicode things), so I'll patch those in a mo too.

closed 2016-07-03 by waveform80

supra2016, 2016-07-14:
Using RPI3; Kernel 4.4.14-v7+ #895, OpenCV 3.1.0, Python 3.4.2.
The PiCamera isn't showing when running on script. I was attempting to do lane detection or speed detection using OpenCV 3.1.0. But it was working before for 5 months. But right now, it doesn't showing anything after I upgrading Kernel 4.4.14.

Here is error msg:
Image width in feet 77 at 76 from camera
Traceback (most recent call last):
  File "/home/pi/opencv31_projects/speed_detector/speed_detector/speed_detector.py", line 131, in <module>
    camera.capture(rawCapture, format="bgr", use_video_port=True)
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1371, in capture
    camera_port, output_port, format, resize, **options)
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 652, in _get_image_encoder
    self, camera_port, output_port, format, resize, **options)
  File "/usr/lib/python3/dist-packages/picamera/encoders.py", line 1049, in __init__
    parent, camera_port, input_port, format, resize, **options)
  File "/usr/lib/python3/dist-packages/picamera/encoders.py", line 534, in __init__
    if not resize and format != 'yuv' and input_port.name.startswith('vc.ril.video_splitter'):
TypeError: startswith first arg must be bytes or a tuple of bytes, not str

Any hint u can help.

waveform80, 2016-07-14:
@supra2016 that sounds like this issue, which was corrected in picamera 1.12. Please check which version of picamera you're using (see [the FAQ](http://picamera.readthedocs.io/en/release-1.12/faq.html#how-can-i-tell-what-version-of-picamera-i-have-installed) for details) - if it's 1.11 you need to upgrade to 1.12.

waveform80, 2016-07-14:
Incidentally, although 1.12 is in PyPI at the time of writing, it's still not hit the Raspbian repositories yet, so upgrading to 1.12 requires a "pip" install (I'll see if I can chase someone about getting "apt" up to date).

supra2016, 2016-07-15:
@waveform80
There are nothing with pi camerea. I been testing with https://www.raspberrypi.org/learning/getting-started-with-picamera/worksheet/
But if I used someone source codes, it will not work.

Unfortunately, I upgraded but doesn't work. I used sudo pip3 install picamera

tsuncp, 2016-07-15:
Sudo pip3 install picamera --upgrade ?

supra2016, 2016-07-15:
@tsuncp
Thank you very much. New version picamera has been upgraded to 1.12

commit 1611ef71 (2016-07-03): Fix #297