waveform80/picamera #299: Unencoded still capture not working in V1.11

blackwoodm, 2016-06-27:
I was getting images at variable resolutions for unencoded capture, up to 1440p in V1.10.  After updating to V1.11, the camera is unable to capture unencoded images, even with the same code.  I used sudo apt-get update && apt-get upgrade to make sure dependencies were up to date, but no dice.

I also tried uninstalling V1.11 and installing V1.10 again, but I got an error stating that it could not find the version.  I tried multiple other versions, but the only one it could identify was 1.11.

I'd like to move back to the previous update until the current one's implementation of unencoded captures is resolved.  How would I go about doing so?

Software/Hardware:
RPi 2
Python3

waveform80, 2016-06-27:
Could you post the code you're using (and the error, if any, that you're getting)?

If you're capturing from the video port and you're just getting black frames - upgrade your firmware to cutting edge (`sudo rpi-update`) and reboot before trying again. The various changes to the pipeline in 1.11 (particularly for video port capture, but still port is also affected to some degree) mean that on older firmwares (and by old I mean a few weeks at the time of writing) you may get completely black frames in certain circumstances (which are rather complex to describe), or an out-right failure to capture.

I'm not sure if it's possible to roll back to old versions of apt installations (I think the Raspbian mirrors only carry the latest version of packages) but you can install any version from PyPI. For example, to uninstall the apt version and install 1.10 from PyPI:

```bash
$ sudo apt-get remove python-picamera python3-picamera
$ sudo apt-get install python-pip python3-pip
$ sudo pip install picamera==1.10
$ sudo pip3 install picamera==1.10
```

blackwoodm, 2016-06-27:
Thanks for the quick reply.  I'm not getting any frame capture, even black frames currently, and my firmware is up to date.

Relevant code:

with picamera.PiCamera() as camera:
                with picamera.array.PiRGBArray(camera) as stream:
                        camera.resolution = (width, height)
                        # max res - 3280 x 2464, causes out of memory error - 1440p works fine
                        
                        camera.start_preview()
                        time.sleep(cameraSleepTime)
                        camera.capture(stream, format='rgb', use_video_port=False, resize=None, splitter_port=0)

Errors:

Traceback (most recent call last):
  File "/home/pi/Camera/practice_ca,mera.py", line 105, in <module>
    camera.capture(stream, format='rgb', use_video_port=False, resize=None, splitter_port=0)
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1371, in capture
    camera_port, output_port, format, resize, **options)
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 652, in _get_image_encoder
    self, camera_port, output_port, format, resize, **options)
  File "/usr/lib/python3/dist-packages/picamera/encoders.py", line 1049, in __init__
    parent, camera_port, input_port, format, resize, **options)
  File "/usr/lib/python3/dist-packages/picamera/encoders.py", line 534, in __init__
    if not resize and format != 'yuv' and input_port.name.startswith('vc.ril.video_splitter'):
TypeError: startswith first arg must be bytes or a tuple of bytes, not str
>>> 

I'm guessing that this has something to do with 'rgb' capture specifically, because of the specific error, but I've not tested 'yuv' yet.  I'll give PyPI a go and see if I get it 1.10 operational again.  I'll post again when I have results.

blackwoodm, 2016-06-27:
Thanks, that worked like a charm.  I was able to install V1.10.  The error disappeared, and the code executed as intended.  I'll continue working on V1.10 for now.

By the way, thank for you this library - it makes camera control far easier.

Jarak-Jakar, 2016-07-01:
I'm having the exact same issue.  I was trying to use recipe 4.2 "Capturing to an OpenCV object" as a basis for a script doing some framerate testing using capture_continuous, and getting an error.  Eventually I worked out that the problem was probably an old version of picamera.  Got that fixed, then started getting this new error.  
[SingleCameraFramerateTest.zip](https://github.com/waveform80/picamera/files/342841/SingleCameraFramerateTest.zip)

To check it wasn't just my code, I tried copy-pasting recipe 4.2 into a new .py and running that.  Same error.  Checked with capturing to JPEG, and that seems to work fine still.

The potentially relevant details I can think of:  RPi firmware is up to date according to rpi-update.  Running Python 3.4.2 - though it is running in a virtual environment, not sure if that will make a difference.  Numpy 1.11.1, though I don't think I was using numpy at all with one of the failing methods in the script.

waveform80, 2016-07-02:
Okay, this is a duplicate of #297 - I'll close this one and continue the discussion there.

closed 2016-07-02 by waveform80