waveform80/picamera #301: PiCameraCircularIO has more seconds than expected

maximedupre, 2016-07-01:
camera = picamera.camera.PiCamera()    
    circularStream = picamera.PiCameraCircularIO(camera, seconds = 10)
    camera.start_recording(circularStream, format = 'h264')

It actually takes me up to 100 seconds to fill up the stream. As you can see, I'm using the default bitrate for both the stream and the camera recording. I'm using `v1.10`. Is that a normal behaviour?

maximedupre, 2016-07-01:
I guess [this answers my question](http://picamera.readthedocs.io/en/release-1.11/faq.html#why-are-there-more-than-20-seconds-of-video-in-the-circular-buffer), but to your knowledge is there a way to have the exact amount of seconds (or +- 0.10 like the duration of a video) in the circular stream?

waveform80, 2016-07-02:
> Actually....I don't want to know the maximum amount of seconds that can fit in a circular stream as it's currently impossible. What I want to know is if there's a way to have a circular stream that accepts a maximum of seconds, rather than a maximum of bytes. I think I got it this time ;).

The number of seconds that'll fit in a stream limited to a number of bytes (including a circular one) is a function of the bitrate of the encoder. The fewer bytes the encoder uses, the more seconds will fit. There's presumably an upper limit on the efficiency of the encoder but I've no idea what it is, so you're absolutely correct that there's no way to specify the maximum number of seconds for a fixed-bytes circular stream at present.

That said, you can specify a minimum number of seconds that will fit given a particular bitrate limit (which is what the `seconds` parameter currently does) and you can then use the [copy_to](http://picamera.readthedocs.io/en/release-1.11/api_streams.html#picamera.PiCameraCircularIO.copy_to) method mentioned in the FAQ you linked to, to copy the last *n* seconds of video from the circular stream.

It would indeed be possible to design a circular stream that accepted a fixed number of seconds, but it would necessarily vary in size (so a naive implementation would be constantly allocating and deallocating memory - a more nuanced one would probably maintain a maximum size and vary its boundaries within that). However, I can't say I've ever seen a use-case for that which wouldn't be fulfilled be the aforementioned `copy_to` method ... yet.

maximedupre, 2016-07-16:
> However, I can't say I've ever seen a use-case for that which wouldn't be fulfilled be the aforementioned copy_to method ... yet.

But the `seconds` param of `copy_to` acts in the same way than the `seconds` param when instantiating `PiCameraCircularIO`, which is to say that it doesn't 

> copy the last n seconds of video from the circular stream

but rather it copies a minimum of the last n seconds. 

I have tried querying the position of the circular stream each second and then taking the bytes that are between the last 10 positions to have a video of a fixed duration and I have been semi-successful.

For some reason, only the odd positions work. For example, If I take the last 5 positions and write the bytes on disk, I successfully get a video with a duration of 5 seconds, but if I take the 6 last positions,  I get a video with a duration of 5 seconds, as opposed to 6. This problem can be extrapolated to any number of seconds (the 8 last positions will give a video duration of 7 seconds). 

I have noticed that there are more bytes that are being written to the circular stream on impair positions (it becomes more noticeable after position 5 (after the camera/encoder has warmed up I suppose)):

Bytes between positions
- **1-2**: 201 999
- **2-3**: 240 072
- **3-4**: 248 720
- **4-5**: 234 004
- **5-6**: 246 758
- **6-7**: 232 370
- **7-8**: 247 195
- **8-9**: 234 044
- **9-10**: 254 012
- **10-11**: 231 309

I don't know if it has anything to do with that, but I have been struggling for days ^^.

Maybe it's an artifact of `ffmpeg` (I wouldn't see why, but we never know).

My script does exactly the same thing for odd or pair positions.

waveform80, 2016-07-16:
>>   However, I can't say I've ever seen a use-case for that which wouldn't be fulfilled be the aforementioned copy_to method ... yet.

> But the seconds param of copy_to acts in the same way than the seconds param when instantiating PiCameraCircularIO, which is to say that it doesn't

>> copy the last n seconds of video from the circular stream

> but rather it copies a minimum of the last n seconds. 

It copies the last *n* seconds, but bear in mind the additional constraint of the *first_frame* parameter which defaults to `sps_header`. This default ensures that you get a valid H.264 stream written (if the stream started with a non-key frame then the stream wouldn't be decodable until the player reached the first key-frame; the reason for using `sps_header` rather than `key_frame` is that a key-frame always follows an SPS header, the header information is tiny by comparison, and potentially useful).

So, by default if you specify `copy_to(seconds=10)` the algorithm seeks backward from  the end of the circular stream looking for the first SPS header frame that is at least 10 seconds from the end. You could specify `copy_to(seconds=10, first_frame=None)` to copy precisely 10 seconds from the circular stream but the result will likely be unreadable until it reaches a key-frame. Of course, you could also specify intra_period=1 when starting the recording which would ensure every single frame is a key-frame but that'll absolutely ruin the compression ratio.

You could also use `request_key_frame` to try and force a key-frame at the position you eventually want to extract from, but that's only a request - there's no way to guarantee a key-frame at a particular position short of recording every frame as a key-frame so there'll also be a degree of imprecision with that.

waveform80, 2017-02-15:
Closing out old tickets - do feel free to re-open if you've got any further questions about this.

closed 2017-02-15 by waveform80