waveform80/picamera #304: Direct capture of Y component to numpy array_Colour image

IgorPotoc, 2016-07-13:
Hello,

I used this program (from Advanced recipes with included matplotlib):

import time, picamera, picamera.array, numpy as np
import matplotlib.pyplot as plt

with picamera.PiCamera() as camera:
    camera.resolution = (100,100)   
    time.sleep(2)
   
    lum_Y = np.empty((112, 128), dtype=np.uint8)     
    try:
        camera.capture(lum_Y, 'yuv') 
    except IOError:
        pass
    lum_Y = lum_Y[:100, :100]   
    print(lum_Y)
    
    plt.imshow(lum_Y)
    plt.show()

How come I got colour image? (I have picamera 1.12 ...)

file:///home/pi/figure_1.png

waveform80, 2016-07-13:
I'd guess you haven't got a colour image; matplotlib's probably applying a default colormap (probably jet if the image looks [ridiculously colourful](http://matplotlib.org/users/image_tutorial.html#applying-pseudocolor-schemes-to-image-plots)) to the Y values. Pick a different colormap (see the `cmap` parameter under that link) like 'greys' and you should see something a bit more sane.

IgorPotoc, 2016-07-19:
Yes, thank you, that was it. I apologize for asking you for some more help here but it's somehow connected (if you like, I can move it to another issue) ... I wanted to compress & decompress this image I captured but I needed uncompressed one to start with (that one is in PNG format). So I saved that numpy array lum_Y to image_Y.data (is this sensible?), like in Advanced Recipes, by adding under plt.show():

with open('image_Y.data', 'wb') as f:
        lum_Y.tofile(f)

Then I wrote a separate code:

import zlib, sys
import numpy as np
import matplotlib.pyplot as plt

unencod = open('image_Y.data','rb').read()   # open saved unencoded image
comprim = open('image_Y_comprim.data','wb')    # open new image file
comprim.write(zlib.compress(unencod,9))     # and write compressed image to it
comprim.close()               

comprim = open('image_Y_comprim.data','rb').read()
decomprim = zlib.decompress(comprim)    # decompress image
decompr = open('image_Y_decomprim.data','wb')   # open new image file
decompr.write(decomprim)        # and write decompressed image to it

print('Size of uncompressed Y:', sys.getsizeof(unencod))
print('Size of compressed Y (9):', sys.getsizeof(comprim))
print('Size of decompressed Y (9):', sys.getsizeof(decomprim))

Y_dec = np.fromfile('image_Y_decomprim.data', dtype=np.uint8)
plt.imshow(Y_dec)
plt.show()

But that produced error:

  File "/home/pi/Zlib_comp_image.py", line 17, in <module>
    plt.imshow(Y_dec)
  File "/usr/lib/python3/dist-packages/matplotlib/pyplot.py", line 2961, in imshow
    imlim=imlim, resample=resample, url=url, **kwargs)
  File "/usr/lib/python3/dist-packages/matplotlib/axes/_axes.py", line 4645, in imshow
    im.set_data(X)
  File "/usr/lib/python3/dist-packages/matplotlib/image.py", line 438, in set_data
    raise TypeError("Invalid dimensions for image data")
TypeError: Invalid dimensions for image data 

Can you suggest how to view the images stored in .data files (I used programs in Advanced Recipes that produced such files, also ...)? I tried with Gimp but I got unrecognizable images. I also tried with numpy.save and load, but as a beginner, I'm obviously doing mistakes that I'm not aware of. Compression and decompression worked (decompressed image is the same size as original and compressed one is smaller) but I would like to visualize decompressed image, too. 
Thank you.