waveform80/picamera #305: Read H264 frame by frame in custom outputs advanced recipe

puneetkp, 2016-07-16:
Hi, 

I am using custom outputs Advanced recipe: [https://picamera.readthedocs.io/en/release-1.12/recipes2.html#custom-outputs](https://picamera.readthedocs.io/en/release-1.12/recipes2.html#custom-outputs). Something like this :
**_camera.start_recording(MyOutput(), format='h264')_**

Inside the write() method of class MyOutput, I would like access the data frame by frame for H264 codec. What would be the fastest way of doing it ?

I could not find any methods in opencv which take the byte stream and grab frames one by one, eg. VideoCapture() works for files and cameras but not on this.
One option I could think of is using the ffmpeg/libavcodec's API **avcodec_decode_video2** but that would require me to write C++ code and then a python binding. 

Any suggestions would be appreciated. Thanks.

waveform80, 2016-07-16:
OpenCV works with BGR format by default. Given that the camera can output BGR directly, why not simply use that and treat each write as an OpenCV image right off the bat (each write will correspond to a single frame; that's *generally* the case with H.264 as well but sometimes key-frames get broken into multiple chunks because they exceed the default recommended buffer size).

puneetkp, 2016-07-16:
I actually want to record a video based on custom processing, the decision whether to record or not is made inside the write() of MyOutput class. 

So, If I use BGR format i won't be able to record video, unless i use the splitter_port (ran into different issue with that mentioned below). Plus the FPS fall way below with Raw captures. For 720p from something like 30+ FPS with H264 to < 5 FPS with Raw BGR capture.

To alleivate this problem I tried using splitter ports something like this:
**camera.start_recording(circularStreamH, format='h264')** for default splitter port = 1
**camera.start_recording(LowResOutput(md), format='mjpeg', splitter_port=2, resize=(160, 120))** for splitter port = 2
In my LowResOutput::write() I decide when to split recording on splitter_port (1), using something like this _camera.split_recording("after.h264")_. But this leads to this error on picamera 1.10:
```
Traceback (most recent call last):
  File "backgnd_subs_live.py", line 375, in <module>
    analyzeStreams()
  File "backgnd_subs_live.py", line 373, in analyzeStreams
    key =cv2.waitKey(1)
  File "/home/pi/.virtualenvs/cv/local/lib/python2.7/site-packages/picamera/camera.py", line 976, in __exit__
    self.close()
  File "/home/pi/.virtualenvs/cv/local/lib/python2.7/site-packages/picamera/camera.py", line 955, in close
    self.stop_recording(splitter_port=port)
  File "/home/pi/.virtualenvs/cv/local/lib/python2.7/site-packages/picamera/camera.py", line 1362, in stop_recording
    self.wait_recording(0, splitter_port)
  File "/home/pi/.virtualenvs/cv/local/lib/python2.7/site-packages/picamera/camera.py", line 1333, in wait_recording
    encoder.wait(timeout)
  File "/home/pi/.virtualenvs/cv/local/lib/python2.7/site-packages/picamera/encoders.py", line 853, in wait
    raise self.exception
picamera.exc.PiCameraRuntimeError: Timed out waiting for a split point
```

waveform80, 2016-07-16:
> So, If I use BGR format i won't be able to record video, unless i use the splitter_port (ran into different issue with that mentioned below). Plus the FPS fall way below with Raw captures. For 720p from something like 30+ FPS with H264 to < 5 FPS with Raw BGR capture.

BGR does indeed go slow, largely because of the huge amount of data involved (uncompressed frames in almost the largest format available). However, if you're going to decode H264 frames with ffmpeg, you're going to encounter the exact same situation because you'll still be throwing uncompressed RGB frames around. You could try and only decode every *n* frames but that'll only work if you stick to key-frames.

> To alleivate this problem I tried using splitter ports something like this:
camera.start_recording(circularStreamH, format='h264') for default splitter port = 1
camera.start_recording(LowResOutput(md), format='mjpeg', splitter_port=2, resize=(160, 120)) for splitter port = 2
In my LowResOutput::write() I decide when to split recording on splitter_port (1), using something like this camera.split_recording("after.h264"). But this leads to this error on picamera 1.10:

That sounds like a much more sensible method (low-res, easily extracted JPEG frames for processing). However, you've run into a threading issue: the `write` method of the custom output is called from the background encoding thread, not from the main script's thread (this must be the case otherwise `start_recording` wouldn't return). The `split_recording` call waits on the background encoding thread to signal that a key-frame has been reached ... unfortunately you're calling `split_recording` from the very thread that it's going to be waiting on, so that's not going to work.

Instead you'll need to set some sort of flag indicating that you wish to split the recording, which the main thread needs to check and act upon. Something like this ought to get you on the right track:

```python
import cv2
from time import sleep
from picamera import PiCamera, PiCameraCircularIO

class AnalysisStream(object):
    def __init__(self):
        self.image_data = b''
        self.state = 'circular'

    def write(self, buf):
        # This runs in the background encoding thread, so you can't
        # start/stop/split recordings here
        if buf.startswith(b'\xff\xd8'):
            if self.image_data:
                self.analyse(self.image_data)
            self.image_data = buf
        else:
            self.image_data += buf

    def analyse(self, image_data):
        # This also runs in the background encoding thread, same rules about
        # start/stop/split recordings. However, we can "communicate" with the
        # main thread via our state variable
        analysis_result = False
        # decode image_data as JPEG and analyse stuff...
        if analysis_result and self.state == 'circular':
            self.state = 'split-to-after'
        elif not analysis_result and self.state == 'after':
            self.state = 'split-to-circular'

with PiCamera() as camera,
    circular_stream = PiCameraCircularIO(camera)
    analysis_stream = AnalysisStream()
    sleep(2) # warm up time
    camera.start_recording(circular_stream, 'h264', splitter_port=1)
    camera.start_recording(analysis_stream, 'mjpeg', splitter_port=2, resize=(160, 120))
    try:
        # This is the main thread's loop. We can start/stop/split recordings
        # here, and we can read what the analysis output wants us to do from
        # the state variable
        while True:
            camera.wait_recording(0.1)
            if analysis_stream.state == 'split-to-after':
                camera.split_recording('after.h264', splitter_port=1)
                analysis_stream.state = 'after'
            elif analysis_stream.state == 'split-to-circular':
                camera.split_recording(circular_stream, splitter_port=1)
                analysis_stream.state = 'circular'
    finally:
        camera.stop_recording(splitter_port=2)
        camera.stop_recording(splitter_port=1)
```

This is one of the reasons the custom outputs section is in the "advanced recipes" section: it necessarily involves dealing with threading issues (I should probably add an explicit note to that effect in the docs).

Also worth noting is that the script above doesn't bother locking the `state` instance variable - it's not actually necessary to lock it in this case because each state change that is checked for can only be driven by the "other" thread. Bear in mind that if you modify the script you may need to start using `threading.Lock` or something similar to eliminate race conditions.

puneetkp, 2016-07-17:
Thanks Dave for your detailed explanation. I tried using the flags for inter thread communication as you described and splitting seems to be working fine. 👍 
I was not aware of the thread situation, now it makes sense. It would be helpful to add threading details in the Advanced recipes, might help others as well.

Talking of the other solution, involving decoding h264 stream:

> However, if you're going to decode H264 frames with ffmpeg, you're going to encounter the exact same situation because you'll still be throwing uncompressed RGB frames around. You could try and only decode every n frames but that'll only work if you stick to key-frames.

I guess bottleneck in RAW frames is due to slow copying of bulky frames from the camera buffer to the opencv process space (My guess so I might be wrong ! ), which would not be the case if decode h264 in my process and then rescale it to smaller size before doing analysis on the frame.
Please share your thoughts on the same. Thanks.

puneetkp, 2016-07-17:
Also in the code that you shared if I try to capture an image at the time of the recording split( from [Capturing images whilst recording](https://picamera.readthedocs.io/en/release-1.12/recipes2.html#capturing-images-whilst-recording) recipe), I see an error like this:

> Traceback (most recent call last):
>   File "backgnd_subs_live.py", line 337, in <module>
>     analyzeStreams()
>   File "backgnd_subs_live.py", line 323, in analyzeStreams
>     camera.capture("split.jpg", use_video_port=True)
>   File "/home/pi/.virtualenvs/cv/local/lib/python2.7/site-packages/picamera/camera.py", line 1579, in capture
>     'Timed out waiting for capture to end')
> picamera.exc.PiCameraRuntimeError: Timed out waiting for capture to end
> 

This is where I added camera.capture():
```
with PiCamera() as camera,
    circular_stream = PiCameraCircularIO(camera)
    analysis_stream = AnalysisStream()
    sleep(2) # warm up time
    camera.start_recording(circular_stream, 'h264', splitter_port=1)
    camera.start_recording(analysis_stream, 'mjpeg', splitter_port=2, resize=(160, 120))
    try:
        # This is the main thread's loop. We can start/stop/split recordings
        # here, and we can read what the analysis output wants us to do from
        # the state variable
        while True:
            camera.wait_recording(0.1)
            if analysis_stream.state == 'split-to-after':
                camera.capture("split.jpg", use_video_port=True, splitter_port=1) # ADDED !
                camera.split_recording('after.h264', splitter_port=1)
                analysis_stream.state = 'after'
            elif analysis_stream.state == 'split-to-circular':
                camera.split_recording(circular_stream, splitter_port=1)
                analysis_stream.state = 'circular'
    finally:
        camera.stop_recording(splitter_port=2)
        camera.stop_recording(splitter_port=1)

```

This one also seems to be related to threading. This time I have called from the main thread, should it not work ?

waveform80, 2017-02-23:
> I guess bottleneck in RAW frames is due to slow copying of bulky frames from the camera buffer to the opencv process space (My guess so I might be wrong ! ), which would not be the case if decode [...]

I'm afraid not - if you decode the H264 in your process you're still having allocate a buffer large enough for the unencoded frames and fill them with data (and that's before you deal with the fact that H264 decoding on the CPU will be horrifically slow compared to the GPU). Basically there's no gain to make here.

> This is where I added camera.capture():
> ...
> This one also seems to be related to threading. This time I have called from the main thread, should it not work ?

You've got a recording running on splitter port 1 (circular_stream), another running on splitter port 2 (analysis_stream), and then during the main loop you're trying to capture on splitter port 1 ... but it's already dealing with a recording. I'm slightly surprised there isn't a more explicit error there (I would've thought that should raise PiCameraAlreadyRecording). Anyway, it should use port 0 or 3 which aren't being used.