waveform80/picamera #306: unable to get a frame with continuous_capture in multiprocessing

AlexisLeC, 2016-07-18:
Hi all,

I can't manage to get a frame with the continuous_capture method while the camera is launched and used in a process. 

`
###################
# Global packages #
###################
import multiprocessing
import logging
import time
# from threading import Thread

###################
# Tierce packages #
###################
import numpy as np
import cv2
from picamera.array import PiRGBArray
from picamera import PiCamera

##################
# Perso packages #
##################

logging.basicConfig(format='%(levelname)s:%(message)s', level=logging.DEBUG)
myLogger = logging.getLogger(__name__)

class CameraWorker(multiprocessing.Process):
    def __init__(self):
        multiprocessing.Process.__init__(self)
        # self.daemon = True
        self._myCamera = Camera()
        return

    def run(self):
        while True:
            self._myCamera.disp_frame()

class Camera(object):
    def __init__(self):
        # cv2.namedWindow('image_display')
        self._camera = PiCamera()
        time.sleep(0.1)
        self._camera.resolution = (640, 480)
        self._camera.framerate = 32
        self._camera.vflip = True
        self._camera.brightness = 50

        self._rawCapture = PiRGBArray(self._camera, size=(640, 480))
        self._stream = self._camera.capture_continuous(self._rawCapture, format="rgb", use_video_port=True)

    def disp_frame(self):
        myLogger.debug("image from rpi cam")
        for f in self._stream:
            myLogger.debug("One frame is getted")
            self._rawCapture.truncate(0)
            # cv2.imshow('image_display', f.array)
            myLogger.debug("stack truncated")
            return True, f.array

if __name__ == "__main__":
    myCamera = CameraWorker()
    myLogger.debug("lets start the camera")
    myCamera.start()

`
Basically nothing output from the "for f in self._stream:" in the disp_frame.
Is it forbidden to have a camera instance in another process ?

reillyardstard, 2016-09-07:
@AlexisLeC - it works fine if you move  ```self._myCamera = Camera()``` out of CameraWorker ```__init__``` and into ```run```  i.e.
```    
    def run(self):
        self._myCamera = Camera()
        while True:
            self._myCamera.disp_frame()
```

waveform80, 2016-09-07:
Weird; has someone recently done a tutorial on using multiprocessing or something? This is about the fifth time in the last couple of months that multiprocessing has come up in conjunction with the camera (#293 and numerous e-mails). I'm going to have to add a FAQ about this ("the PiCamera object can only be used in a single process, share it and stuff breaks" or something along those lines.

Actually, I may as well make this the ticket to remind me!

AlexisLeC, 2016-09-07:
@reillyardstard Thanks, I will try that

commit 05281b20 (2016-11-15): Fix #306
closed 2016-11-15 by waveform80