waveform80/picamera #307: Dropped frames with PiCameraCircularIO, two splitter ports with 'resize'

slackista, 2016-07-21:
First of all big thanks for great job with picamera!
I play with motion detection based on two camera streams, one for motion detection (numpy array) and second to write video while motion is detected:
```
with picamera.PiCamera() as camera:
   camera.resolution = (1296, 972)
   camera.framerate = 15

camera.start_recording(stream, format='h264', splitter_port=1, inline_headers=True, intra_period=video_intra_period)
   camera.start_recording('/dev/null', splitter_port=2, format='h264', motion_output=MyMotionDetector(camera, size=(1296,972)))
```

Everything is fine, no dropped frames at all. Cpu usage~8%. Even now when I've updated picamera 1.10>1.12 the error "Timed out waiting for spilt point" didn't occur.

Because this is binned resolution, lines on video are chopped an quality isn't great, I would like to capture with full frame, resize in splitter port 1 for motion detection and resize for write video in splitter port 2:

```
with picamera.PiCamera() as camera:
   camera.resolution = (2592, 1944)
   camera.framerate = 15

camera.start_recording(stream, format='h264',resize=(1296, 972), splitter_port=1, inline_headers=True, intra_period=video_intra_period)
   camera.start_recording('/dev/null', splitter_port=2, format='h264', resize=(1296, 972), motion_output=MyMotionDetector(camera, size=(1296,972)))
```

There are a lot of frames dropped, I can calculate in average video is around 6 fps. Cpu usage is around 10%. I've tried to overclock GPU, with gpu_freq=500 but it's almost the same. Sometimes script hangs and camera resources are unavailable unttill reboot.
My code is based on: [https://gist.github.com/spikedrba/aecb82d8b51e991dbd01] My setup: RPI3, camera v1, gpu_mem=512.

6by9, 2016-07-22:
Using the resize option is a resize on the GPU vector processor - it is not hardware accelerated and there is a finite amount of processing power there. Reading in 7.5MB for each image (5MPix YUV420) processing, and writing 1.8MB for 1.25MPix ouput is not a trivial task, and you're doing it twice.

If you really think the binning is giving image quality issues, then use ```sensor_mode=2``` (see http://picamera.readthedocs.io/en/release-1.12/api_camera.html#picamera.PiCamera.sensor_mode) to force the 5MPix mode off the sensor, and then ask for 1296x972 as the camera output resolution. The resize is then done once in the ISP hardware block and fed to the two outputs. That should keep up at 15fps fairly happily.

slackista, 2016-07-22:
Indeed one resizing operation should do the trick, but how ? What do you exactly mean "ask for 1296x972" ?

6by9, 2016-07-22:
Python is not a language I have much expertise in, but something like
```
with picamera.PiCamera(sensor_mode=2) as camera:
   camera.resolution = (1296, 972)
   camera.framerate = 15
   camera.start_recording(stream, format='h264', splitter_port=1, inline_headers=True, intra_period=video_intra_period)
   camera.start_recording('/dev/null', splitter_port=2, format='h264', motion_output=MyMotionDetector(camera, size=(1296,972)))
```
So the same as your first example where you complain about binning, but set sensor_mode to force the mode to 5MP off the sensor.
There are some intricacies around when and how to set sensor_mode for it to work - read the docs, but as early as possible is good.

slackista, 2016-07-22:
Strange:
```
import` picamera

camera = picamera.PiCamera(sensor_mode = 2)
camera.rotation = 180
camera.start_recording('my_video.h264')
print(camera.sensor_mode)
camera.wait_recording(5)
camera.stop_recording()
```

camera.sensor_mode object retrieves '0' no matter what I set in constructor. Furthermore according to wiki this script should fail hance h264 limitation. Instead of that it is working and video output is 720x480. 

```
import picamera

camera = picamera.PiCamera(sensor_mode = 2)
camera.resolution = (1296, 972)
camera.rotation = 180
camera.start_recording('my_video.h264')
print(camera.sensor_mode)
camera.wait_recording(5)
camera.stop_recording()
```
This time 'camera.sensor_mode' retrievers 0 and video is binned 1296x972. Somehow sensor_mode parameter is omitted.

slackista, 2016-07-22:
I think there is a bug in latest version of PiCamera. When downgraded to v1.10 everything works fine, camera.sensor_mode returns previous set value  and video output is correct. Should I open new ticked for this ?

6by9, 2016-07-22:
https://github.com/waveform80/picamera/blob/master/picamera/camera.py#L1977
```
    def _configure_camera(
            self, sensor_mode, framerate, resolution, clock_mode,
            old_sensor_mode=0):
...
        if old_sensor_mode != 0 and sensor_mode != 0:
            self._camera.control.params[mmal.MMAL_PARAMETER_CAMERA_CUSTOM_SENSOR_CONFIG] = sensor_mode
```

looks like some odd logic, as it means that the call from https://github.com/waveform80/picamera/blob/master/picamera/camera.py#L416 will never actually set the mode. Perhaps I'm missing something that happens later on, but it looks like that initialiser value is discarded.

It vaguely looks like setting the property after initialisation should work.
Perhaps this is why http://picamera.readthedocs.io/en/release-1.12/api_camera.html#picamera.PiCamera.sensor_mode has the comment
> At the time of writing, setting this property does nothing unless the camera has been initialized with a sensor mode other than 0. Furthermore, some mode transitions appear to require setting the property twice (in a row). This appears to be a firmware limitation.

I'm not aware of any firmware limitations along those lines, although it's not a value that is expected to change dynamically.

waveform80, 2016-07-23:
> looks like some odd logic, as it means that the call from https://github.com/waveform80/picamera/blob/master/picamera/camera.py#L416 will never actually set the mode. Perhaps I'm missing something that happens later on, but it looks like that initialiser value is discarded.

The main reason for the weird `if` statement there is support for old firmwares that lack the ability to set the sensor mode directly. The intent is that, if the old sensor mode is the default (0 or auto) and the new sensor mode is also 0, we never attempt to set the CUSTOM_SENSOR_CONFIG parameter (which would fail on old firmwares that don't support it). However, if a custom sensor mode has been set (`old_sensor_mode` is not 0) or if the user is explicitly requesting a custom sensor mode (`sensor_mode` is not 0) then we set the parameter and hope the firmware's recent enough to accept it (which it almost certainly is given that parameter's been supported for years).

However, you're absolutely right: the test is incorrect. Should be `not (old_sensor_mode == 0 and sensor_mode == 0)` or `old_sensor_mode != 0 or sensor_mode != 0`. Ooops.

> I'm not aware of any firmware limitations along those lines, although it's not a value that is expected to change dynamically.

I vaguely recall playing with it manually on the command line and finding some odd effects when changing it dynamically (like the repeat setting thing) - but as you say, it's pretty obviously one of those parameters that the firmware probably expects to be set soon after init and then never touched again (and I can't imagine that many use cases for changing it dynamically - if that's required then the auto setting should generally be preferable). Still, I figured since it can be set dynamically, might as well throw it in there for anybody that desperately wants it.

> I think there is a bug in latest version of PiCamera. When downgraded to v1.10 everything works fine, camera.sensor_mode returns previous set value and video output is correct. Should I open new ticked for this ?

Yup ... it's a bug. Do feel free to open a ticket (or even a PR :).

closed 2017-02-23 by waveform80