waveform80/picamera #311: Issue with capture_continuous

weathernick, 2016-08-06:
I think I found a bug or a feature?

I run capture_continuous and have it pause for 6ish seconds. I just noticed when I upgraded to 1.12 that back to back images are the same. So for time 0sec I have an images, 6 seconds later I grab another image from capture_continuous and it turns out to be the same image. 6 seconds later I get a new image but it gets doubled on the next cycle. so instead of ABCDEFG I get AACCEEGG. 

My code: 
`   
    for i, filename in enumerate(camera.capture_continuous(nightpath+'/{timestamp:%H%M%S}.jpg', resize=resolution, burst=True, quality=quality)):
        sleep = nextpictime("night",filename)
        time.sleep(sleep)
`

I tried turning off burst and addjusting my sleep but same results. I reverted back to 1.10 which works fine for me and my four weather cameras. 

Just figured I would let you know. 

Thanks,

Nickolas

waveform80, 2017-02-23:
Hmm, that took some work but I've got this fixed up now. Many thanks for mentioning which version worked - that helped a great deal (although unfortunately a *huge* amount changed under the covers from 1.10 to 1.11 which took some time to dig through - anyway, it was indeed one of those changes that caused the issue).

closed 2017-02-23 by waveform80

alexellis, 2019-03-15:
I feel like I'm seeing the same issue with this code:

```
  engine = DetectionEngine(args.model)
  labels = ReadLabelFile(args.label) if args.label else None

  with picamera.PiCamera() as camera:
    camera.resolution = (640, 480)
    camera.framerate = 30
    _, width, height, channels = engine.get_input_tensor_shape()
    camera.start_preview()
    try:
        stream = io.BytesIO()
        for foo in camera.capture_continuous(stream,
                                                format='jpeg',
                                                use_video_port=True,
                                                resize=(width, height)):
            # stream.truncate()
            stream.seek(0)
            # input = np.frombuffer(stream.getvalue(), dtype=np.uint8)
            img = Image.open(stream)
            
            # w, h = 640, 480
            # data = np.zeros((h, w, 3), dtype=np.uint8)
            # data[256, 256] = [255, 0, 0]
            # img = Image.fromarray(input, 'RGB')

            ans = engine.DetectWithImage(img, threshold=0.1, keep_aspect_ratio=True,
                                            relative_coord=False, top_k=10)
            img=None
            print("ans", ans)     
            if ans:
                for obj in ans:
                    print ('-----------------------------------------')
                    if labels:
                        print(labels[obj.label_id])
                        print ('score = ', obj.score)
    finally:
        camera.stop_preview()
        stream.close()
```

I can only get different classifications if I close the program and re-open it.

What would you advise? I appear to have "picamera 1.13"

Alex

commit c4ffce9c (2017-02-23): Fix #311

On MMAL ports, after the buffer is processed and released, always get a
buffer and send it to the port. Originally, mmalobj tried to be "nicer"
about this; waiting for a buffer and not sending anything if the handler
has signalled it's finished (to prevent further data flow) but this
seems to cause issues with burst captures (images get repeated either
because something is still left in the pipeline, or because some buffer
gets resent - not sure which, but this fixes the issue)