waveform80/picamera #312: Duplicate entry in hardware table doc

thijstriemstra, 2016-08-07:
I see this entry twice for the v2 camera:

```
+---+------------+--------------+------------+-------+-------+---------+---------+
| 2 | 3280x2464  | 4:3          | 0.1-15fps  | x     | x     | Full    | None    |
+---+------------+--------------+------------+-------+-------+---------+---------+
| 3 | 3280x2464  | 4:3          | 0.1-15fps  | x     | x     | Full    | None    |
+---+------------+--------------+------------+-------+-------+---------+---------+
```

waveform80, 2016-08-07:
Yup, that's intended - the V2 camera module's firmware modes were adjusted to match the V1 module's modes as closely as possible.

While the V1's two still modes were split between "slow framerates" (<1fps) and "normal framerates" (>=1fps), the V2's single still mode covered all possible framerates (0.1 all the way up to 15fps). In order to match the V1's modes more closely, that single mode was duplicated. This means that if you have some code which specifically selects [sensor_mode](http://picamera.readthedocs.io/en/release-1.12/api_camera.html#picamera.PiCamera.sensor_mode) 3 for slow framerates it'll operate more or less the same on both the V1 and V2 modules.

thijstriemstra, 2016-08-07:
Thanks for the explanation. Perhaps it's useful to add a concise version of this note the docs, maybe with an asterisk behind 3, e.g. 3*? I can make a PR.

thijstriemstra, 2016-08-07:
ps. while developing my app I built a quite extended control panel for picamera's settings using PyQt, maybe this could be included in picamera (maybe as an example app) or should this live outside this repository?

waveform80, 2016-08-08:
> Thanks for the explanation. Perhaps it's useful to add a concise version of this note the docs, maybe with an asterisk behind 3, e.g. 3*? I can make a PR.

Yeah, that's probably worth a note - if you fancy making a PR, great!

> ps. while developing my app I built a quite extended control panel for picamera's settings using PyQt, maybe this could be included in picamera (maybe as an example app) or should this live outside this repository?

I wouldn't be keen on adding something heavy like PyQt/PySide as a dependency (I was fairly reticent about adding numpy as a dep, which is why it's an option and `picamera.array` isn't pulled into the main module by default). Also, PyQt used to be horrid to install via pip (last time I tried, admittedly a year ago or so), hence I'm none to keen to add it even as an option (just thinking about all the extra support tickets such a thing can generate :).

Still, a GUI is probably interesting to numerous picamera users so I'm happy to add a link to it from the docs (I should probably add a section for "interesting projects" as there's plenty out there).

commit 1f03354a (2017-02-23): Fix #312

Added a note about the apparent duplication of sensor mode 3 on the V2
module
closed 2017-02-23 by waveform80