waveform80/picamera #313: Re-using PiBayerArray

rwb27, 2016-08-08:
I used a `picamera.array.PiBayerArray` to store the raw image from my camera, and then to convert it to a more manageable RGB format - it's very convenient so thanks for including this.

I used this in a loop to acquire lots of images and it crashes after acquiring about 62; I think this is because I didn't call `truncate(0)` each time.  Would it make sense to automatically truncate the array when it's used for a new image?  I'm happy to have a go at some code to do that if it would be helpful.

Thanks,

Richard

waveform80, 2016-08-08:
Hmmm. Technically, that's a backwards incompatible change - still, I'm struggling to think of a use-case where someone would actually want to store multiple images in the stream, so it's worthwhile considering.

However, at this point I'm considering deprecating the various Pi*Array classes anyway, because [direct capture to a numpy array](http://picamera.readthedocs.io/en/release-1.12/recipes2.html#capturing-to-a-numpy-array) is much more convenient (doesn't require messing around with stream truncation as you're just overwriting the array data). That'll require re-writing most of the advanced recipes so they don't mention the old stream-based classes. I'm just debating what to do about the added functions like YUV->RGB conversion, Bayer->RGB, and so forth. Utility functions? Another class? Not sure what's easiest for users yet.

rwb27, 2016-08-09:
Yes, the backwards-compatibility thing did occur to me, I suppose it would be better to avoid that.  Perhaps you could add utility functions but allow users to access it with different format arguments (e.g. format="yuv->rgb" or format="bayer->rgb")?  That would seem quite neat, though opinions may vary!

I have been looking at the code a bit more, and I think I might produce a quicker-and-dirtier hack of your demosaicing code for my purposes (I'm using the camera in a microscope, so the resolution of the optics is much less than the number of pixels).  I think it should be possible to speed things up a lot by combining demosaicing with binning - and only keeping either the top or bottom 8 bits of the image.

Anyway, I suspect I should close this, as it's not really a bug - I just wanted to mention it.  Thanks for taking the time to reply :)

closed 2016-08-09 by rwb27