waveform80/picamera #316: Correction on circular buffer example code

emmanuel-contreras, 2016-08-17:
Hello,
As I was trying the code on "3.12. Recording to a circular stream" I noticed that it kept recording everything since the stream started, into the file, not the seconds since the motion event trigger was detected. It wasn't until I looked at the issue below that I realized that seconds was missing from the example "copy_to('motion.h264', seconds=10)". Now it seems to be recording more or less the correct number of seconds (35s vs 30 specified) Should that correction be made on the example?

http://picamera.readthedocs.io/en/release-1.12/recipes1.html#recording-to-a-circular-stream
https://github.com/waveform80/picamera/issues/301

waveform80, 2016-08-17:
No, it won't be *everything* since the stream started because the stream is circular. However, take note of the text at the end of the recipe:

> Note that *at least* 20 seconds of video are in the stream. This is an estimate only; if the H.264 encoder requires less than the specified bitrate (17Mbps by default) for recording the video, then more than 20 seconds of video will be available in the stream.

The default bitrate is extremely generous (it's intended for high quality 1080p30 recording - if you're using a lower resolution and there's little motion you'll find *much* more than 20 seconds of data in the stream (it's entirely possible to store several minutes within the buffer).

I was tempted to add the new `seconds` parameter to the `copy_to` call in that recipe, but I think it's more important for people to learn that the video isn't fixed-rate and thus that the circular buffer's seconds length is a lower bound only (that's also why I added [this FAQ](http://picamera.readthedocs.io/en/release-1.12/faq.html#why-are-there-more-than-20-seconds-of-video-in-the-circular-buffer) as this does get asked quite a bit).

emmanuel-contreras, 2016-08-17:
Thank you so much for the explanation, that all makes sense, and I learned something new today :)

closed 2016-08-17 by emmanuel-contreras