waveform80/picamera #317: Q: Is there looping video recording?

nickwebcouk, 2016-08-20:
Is there an easy recipients to enable loop recording? I.e set a video time of 60 seconds and picamera will continually create new files named n.h264 for every minute (or thereabouts with key frames) just like dash cams do? And the possibility of deleting the oldest when either you reach a certain sequence number or it detects that space is low on whatever drive is being used?

Sorry if this is in the docs - I couldn't see it and I had a look through the advanced recipes. 

Cheers for any help!

Nick

waveform80, 2016-08-20:
[record_sequence](http://picamera.readthedocs.io/en/release-1.12/api_camera.html#picamera.PiCamera.record_sequence) is the basis of what you want but requires some additional programming from you. There's some basic examples in [this recipe](http://picamera.readthedocs.io/en/release-1.12/recipes1.html#recording-over-multiple-files) but I'm sure I did some more advanced ones somewhere (not in the docs ... some forum or something)... Just can't seem to find them at the moment...

waveform80, 2016-08-20:
Ah, found them: check out the end of [this answer](http://raspberrypi.stackexchange.com/questions/26714/can-i-record-a-24-hour-video-on-the-raspberry-pi-with-camera-module/26746#26746) on the RPi stackex.

paulhothersall, 2016-08-21:
ahem, if you are recording a live loop, then I HIGHLY recommend a ram disk if you are using actual files for the buffer.

Your SD card will thank you later.

waveform80, 2016-08-21:
If you just need a circular buffer, [PiCameraCircularIO](http://picamera.readthedocs.io/en/release-1.12/api_streams.html#picameracirculario) is what you're looking for, which is indeed RAM based. However, I get the impression this is about long term buffering of recorded data (several days worth if deletion is only to occur when low on space), in which case RAM is going to be far too limited in capacity (in the Pi3's 1Gb of RAM, with decent compression levels, one can store several minutes worth of video at 24fps, but that's about it).

nickwebcouk, 2016-08-21:
in essence, yes. I'm hoping to use picamera as the basis for a dash cam project. The script will start recording at a sequence number (either based on date or the last sequence number) and will record to pre-defined chunks of time (commonly 5 minutes, but 1 minute to two minutes is also common). When space is low on whatever medium is being used to save the file, the oldest is deleted, and it continues perpetually (well, until it's switched off or recording stops etc). I'm hoping to write directly to the medium and not use a ton of ram.

I know raspivid has the ability by way of the segment, wrap and start command line options (https://www.raspberrypi.org/documentation/raspbian/applications/camera.md). Is there anything like this expected in picamera? I know its quite specific but if you dont ask and all that jazz!

Thanks for the comments so far - It's definitely something i'm going to have a play around with and see what I can figure out.

As a side note - is there any official forums where questions like this can be asked instead of clogging up the bug tracker?

Thanks agian!

Nick

waveform80, 2016-08-21:
> I know raspivid has the ability by way of the segment, wrap and start command line options (https://www.raspberrypi.org/documentation/raspbian/applications/camera.md). Is there anything like this expected in picamera? I know its quite specific but if you dont ask and all that jazz!

Nothing that specific, because that's not very useful in a library (making a library's function too specific tends to make it hard / impossible to implement things that don't fall neatly into that specification). Far more useful to provide a generic mechanism. In this case, to provide a mechanism which automatically handles rolling recordings over from one output to another, taking care to ensure that recordings always start with the appropriate headers: that's exactly what [record_sequence](http://picamera.readthedocs.io/en/release-1.12/api_camera.html#picamera.PiCamera.record_sequence) does.

At that point all you need to code is the criteria indicating when to roll over to the next recording (which occurs when you permit the loop to iterate) - which can be anything you could imagine (and code): time based, space based, reaction to a button, a sensor, receiving a particular network packet, etc. etc.

> As a side note - is there any official forums where questions like this can be asked instead of clogging up the bug tracker?

There's the [Raspberry Pi camera board](https://www.raspberrypi.org/forums/viewforum.php?f=43) - unfortunately I don't have enough spare time to respond on there as I used to, as well as the [Pi Stack Exchange](http://raspberrypi.stackexchange.com/). Still, I've no objection at all to people asking questions here. It's a bit limited as forum interfaces go (hardly surprising given it's not exactly its intended purpose), but it's still public and searchable which means any answers can still be of use to anyone else googling for them.

waveform80, 2017-02-04:
Closing old issues; I think this one's been answered adequately but do feel free to re-open if you've got any further queries regarding it!

closed 2017-02-04 by waveform80