waveform80/picamera #318: Can not set awb_gains

messismore, 2016-08-24:
Hi everyone,

I am trying to hard code a certain white balance for a time lapse camera, however, it seems like setting `camera.awb_gains = Fraction(497, 256), Fraction(177, 128)` has no effect. The code below outputs a different awb_gains each time.

```
import picamera
import time
from fractions import Fraction

camera = picamera.PiCamera()
camera.framerate = 10

time.sleep(1)
# This capture discovers initial AWB and SS.
camera.capture('try.jpg')
camera.shutter_speed = camera.exposure_speed
currentss = camera.exposure_speed
camera.exposure_mode = 'off'
camera.resolution = (1920, 1080)
camera.awb_mode = 'off'
camera.awb_gains = (Fraction(497, 256), Fraction(177, 128))

print camera.awb_gains
```
```
$ python wb_gains_test.py 
(Fraction(203, 256), Fraction(675, 256))
$ python wb_gains_test.py 
(Fraction(211, 256), Fraction(367, 128))
$ python wb_gains_test.py
(Fraction(197, 256), Fraction(675, 256))
```

What a I doing wrong?

Berryfier, 2016-10-21:
The value of awb_gains used by print seems to take a while to settle. Inserting repeat print and capture statements will reveal this unexpected behaviour. However, looking at repeat captures, the used value of awb_gains does seem to be very close to what is input.

waveform80, 2016-11-14:
This looks pretty comprehensively answered to me, so I'll close it (feel free to re-open if you've got anything further to add though!)

closed 2016-11-14 by waveform80

JolleJolles, 2019-07-31:
I have a similar issue. I created the following script based on the picamera documentation

`from time import sleep
from picamera import PiCamera
from fractions import Fraction

camera = PiCamera(resolution=(1280, 720), framerate=30)
# Set ISO to the desired value
camera.iso = 100
# Wait for the automatic gain control to settle
sleep(2)
# Now fix the values
camera.shutter_speed = camera.exposure_speed
camera.exposure_mode = 'off'
g = camera.awb_gains
camera.awb_mode = 'off'
camera.awb_gains = (2.5, 2.5)
# Finally, take several photos with the fixed settings
camera.capture('try.jpg')
print(camera.awb_gains)`

But whatever I put as `camera.awb_gains`, also with Fractions, the values returned are always very similar fractions in the range `(Fraction(167, 128), Fraction(257, 128))`. The images returned also always have a nice white balance and not the overly red or blue that I would expect with the provided awb_gains.