waveform80/picamera #320: failed to get a buffer from the pool

friedcpu, 2016-09-04:
I am looking for a solution to this problem, I am sorry I can't offer any insight into why it is happening.

Error:
```
line 66, in main
    overlay_renderer.update(img.tostring())
  File "/usr/local/lib/python2.7/dist-packages/picamera/renderers.py", line 401, in update
    buf = self.renderer.inputs[0].pool.get_buffer()
  File "/usr/local/lib/python2.7/dist-packages/picamera/mmalobj.py", line 1185, in get_buffer
    raise PiCameraRuntimeError('failed to get a buffer from the pool')
PiCameraRuntimeError: failed to get a buffer from the pool
```

Code:
```
overlay_renderer = None
  while True:
    text = "TESTING"
    img = Image.new('RGB', (1024, 768))
    draw = ImageDraw.Draw(img)
    draw.font = ImageFont.truetype("/usr/share/fonts/truetype/freefont/FreeSeri$
    draw.text((10,10), text, (255, 0, 255))

    if not overlay_renderer:
      overlay_renderer = camera.add_overlay(img.tostring(), layer=3, size=img.s$
    else:
      overlay_renderer.update(img.tostring())
```

I have tried other peoples code that use add_overlay and it appears if it isn't static and requires a loop to update then it fails to get a buffer on the 2nd loop.

Thanks for your help

waveform80, 2016-09-04:
Ah, this is an interesting one! It's not actually failing on the 2nd loop - it's failing on the 4th which if you know a bit about MMAL's buffer defaults is a great big hint as to what's going on.

The short answer is: throw a sleep into your loop which keeps it slower than the camera's framerate and you'll be fine.

The long answer:

There's quite a lot of stuff going on "behind the curtain" here. Overlays are implemented as MMAL renderer components (MMAL being the library underlying picamera). Naturally these components have an input port used to feed the renderer images to display. Each input (or output) port in MMAL is associated with a pool of buffers which are used to pass data around (in this case each buffer will represent an image to be displayed by the renderer).

By default, MMAL recommends that the pool is initialized with 3 buffers (picamera generally follows MMAL's recommendations). I presume this is so the renderer component can have one buffer to work on displaying, the client application can have another one to work on prepping the next frame, and a third for a bit of leeway.

Anyway, that's why it's failing on the 4th iteration of the loop: the pool has been exhausted of its 3 buffers and the fourth `update` call throws an exception as a result. Why is the pool exhausted? Or more precisely, why haven't buffers been returned to the pool by the renderer? Because the renderer hasn't had a chance to display them yet. The renderer is running at the framerate of the camera (I think? I'm not actually sure about that bit ... might be the display framerate). Anyway, it's not running as fast as your loop, so your loop needs to be slowed down to give the renderer a chance to display the image you're sending it and return the buffer containing that image to the pool.

Here's a quick variation of your code which works happily; remove the `sleep` to see it break :)

```python
from picamera import PiCamera
from PIL import Image, ImageDraw, ImageFont
from time import sleep

camera = PiCamera(resolution='1024x768', framerate=30)
camera.start_preview()
overlay_renderer = None
i = 0
while True:
    text = "TESTING %d" % i
    i += 1
    img = Image.new('RGB', (1024, 768))
    draw = ImageDraw.Draw(img)
    draw.font = ImageFont.truetype("/usr/share/fonts/truetype/freefont/FreeSerif.ttf")
    draw.text((10,10), text, (255, 0, 255))

    if not overlay_renderer:
        overlay_renderer = camera.add_overlay(img.tostring(), layer=4, alpha=128, size=img.size)
    else:
        overlay_renderer.update(img.tostring())
    sleep(0.1)
```

closed 2016-09-29 by waveform80

6by9, 2017-08-31:
(Only just seen this one via the linking to 436)
To answer waveform80's questions, the renderer will be running at the display refresh rate, so typically 60Hz.
It double buffers, so three buffers is normally one on the display, one waiting to be displayed on the next vsync, and one with the client (picamera) being filled. Supplying a new update before the vsync will result in the previously queued buffer to be returned. On the vsync the buffer that was on the display will be returned.

I suspect that it is more a scheduling issue that the MMAL callback releasing the buffer back into the buffer pool hasn't had a chance to be scheduled before the next attempt, but whichever way there is absolutely no point in continuously firing buffers into a renderer as the supplied code was. 60Hz is the max that makes any sense (unless you are playing back high frame rate video and deliberately wanting to drop frames).