waveform80/picamera #322: Add method to flush macro-blocks to file immediately.

PiStuffing, 2016-09-06:
I'm using macro-blocks for motion tracking; They are written to a FIFO.  However, they are buffered by the OS, so that only when the macro-blocks exceed 65536 bytes are they flushed by the OS into the FIFO.

My setup is 330x320 pixels at 10fps => 21 x 20 macro-blocks x 4 bytes = 1680 bytes per frame of macro-blocks.  These are buffered up and only written into the FIFO roughly every 4 seconds (i.e. 1680 x 10Hz * 4s > 65536 bytes ).

Raspivid doesn't have this buffering, so I assume it flushes each frame of macro-blocks directly to file.

Would it be possible to add an equivalent to picamera?

Thanks

6by9, 2016-09-06:
It helps to quote the forum discussion on this - https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=159115

raspivid has AN OPTION to call flush after every write. This has a performance implication when actually writing to a file system, hence why it isn't the default. If piping it to a network socket or Linux pipe, then not flushing can increase latency.

I'm guessing (my Python really is bad) that we want an optional flush around https://github.com/waveform80/picamera/blob/master/picamera/encoders.py#L304, or we appear to be able to change the mode at open.

PiStuffing, 2016-09-06:
Quoting from an e-mail by waveform80, no fix is required:

So, in your case instead of this:

camera.start_recording(
    '/dev/null', format='h264',
    motion_output="/dev/shm/motion_stream", quality=23)

You want something like this (this assumes Python 3's variant of open; if you're on Python 2, use the "io" library's open() function instead):

output = open('/dev/shm/motion_stream', 'wb', buffering=0)
try:
    camera.start_recording(
        '/dev/null', format='h264', motion_output=output,
        quality=23)
    # wait_recording, etc.
finally:
    camera.stop_recording()
    output.close()

Basically, when people give picamera a filename (as opposed to a file-like object), I have to make a judgement call and open the file in a way that fits the majority use case. The majority use case is "I don't want to bash every minor write onto the SD card and drastically reduce its performance and lifespan", so the default has to include a fairly substantial amount of buffering. That's the logic behind the default.

However, there's plenty of perfectly good use cases (like yours) in which buffering is a problem - hence why you can pass any file-like object to picamera and it'll happily write to it like any other output.

closed 2016-09-06 by PiStuffing

waveform80, 2016-09-06:
Sorry, hadn't noticed this (been buried in gpio-zero the last few days).

While I don't think anything needs adding to the library, I'm tempted to make this a docs enhancement. I don't think I've mentioned anywhere that picamera opens files with an extra-large buffer by default (Python's default buffer is the OS block size - 4k, generally I think; picamera deliberately uses a larger 64k because the vast majority of our IO is sequential so there's a reasonable benefit). It's probably worth mentioning at least in the API docs, if not in the FAQ.