waveform80/picamera #324: Setting 'sensor_mode' has no effect

dneunkirch, 2016-09-11:
After the upgrade from version 1.09 to 1.12 has setting the `sensor_mode` no effect anymore (same behavior on 6 pi's). 

Example:

```
#!/usr/bin/python
import picamera

with picamera.PiCamera(sensor_mode=2, resolution=(1920, 1080), framerate=15) as camera:
  camera.capture('/var/www/html/preview.jpg', format='jpeg', quality=10, use_video_port=True, splitter_port=2)
```

If I take a picture with `raspistill` it's working (`raspistill -w 1920 -h 1080 -md 2 -o test.jpg`).

Problem on:
`4.4.20-v7+ #908 SMP Wed Sep 7 14:44:27 BST 2016 armv7l GNU/Linux` (Raspberry 3)
`4.4.20+ #908 Wed Sep 7 14:35:04 BST 2016 armv6l GNU/Linux` (Raspberry B, B+, 2)

dneunkirch, 2016-09-14:
After further analysis I have found out that the problem comes with the release 1.11.

If you want to get the current sensor_mode  with `camera.sensor_mode` it always returns `0` but the method `configure_camera(...)` is correctly invoked and sets the given sensor_mode. It seems so that the parameter `MMAL_PARAMETER_CAMERA_CUSTOM_SENSOR_CONFIG` is wrong, but on the same firmware with release 1.10 it's working.

waveform80, 2016-09-29:
Yup, confirmed - I'll have a look into this tonight.

waveform80, 2016-09-29:
Doh, I *knew* this rang a bell - 6by9 pointed it out back in #307 and I completely forgot about it!

commit f0f2f23e (2016-09-29): Fix #324
closed 2016-09-29 by waveform80