waveform80/picamera #325: Flash GPIO output on RPi 3

rtolesnikov, 2016-09-12:
The following describes the process to get a discrete GPIO to indicate when the image is taken. However, there is no device tree mentioned for RPI3. Is anyone able to making this happen on RPI3?

https://github.com/waveform80/picamera/blob/master/docs/recipes2.rst

6by9, 2016-09-12:
Docs need to be updated to point at https://github.com/raspberrypi/firmware/blob/master/extra/dt-blob.dts
Platform names also changed ages back, so the platform to block mapping needs to be:
Pi Zero - pins_pi0
Pi B rev1 - pins_rev1
Pi B rev 2 or A - pins_rev2
Pi B+ rev1.1 - pins_bplus1
Pi B+ rev1.2 - pins_bplus2
Pi A+ - pins_aplus
Pi 2 rev1 - pins_2b1
Pi 2 rev1.1 - pins2b2
Pi 3 rev1.0 (never released?) - pins_3b1
Pi3 rev 1.2 - pins_3b2

waveform80, 2016-09-12:
Ah, looks like some more doc updates are required! Thanks for the info :)

6by9, 2016-09-12:
There are a few updates for you (eg your wget line), but also https://www.raspberrypi.org/documentation/configuration/pin-configuration.md needs to be updated to point to the correct place.
Ah https://github.com/raspberrypi/documentation/issues/359 - I thought it had been logged already.

commit a1a60a0a (2016-11-15): Fix #325
closed 2016-11-15 by waveform80