waveform80/picamera #326: picamera image capture not working

raymondodinzeo, 2016-09-12:
When I run 
#!/usr/bin/python
from time import sleep
import picamera

with picamera.PiCamera() as camera:

    camera.start_preview()
    #camera.start_recording('foo.h264')
    #camera.wait_recording(60)
    sleep(2)
    camera.capture('foo.jpg')
    camera.stop_preview()

I get
<module 'time' (built-in)>

Traceback (most recent call last):
  File "/home/pi/piccapture.py", line 11, in <module>
    camera.capture('foo.jpg')
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 1570, in capture
    encoder.start(output)
  File "/usr/lib/python2.7/dist-packages/picamera/encoders.py", line 1529, in start
    timestamp = datetime.datetime.now()
AttributeError: 'module' object has no attribute 'datetime'
>>> 

If I comment out "camera.capture('foo.jpg')" 
the code runs with no errors.
If I do a video capture everything works ok.

Please help!

looking at problems other people have had I have just run sudo apt-get dist-upgrade to make sure everything is up to date. Still no joy.

waveform80, 2016-09-13:
Would you happen to have a file named "datetime.py" somewhere in the same directory? If so: that's the issue - you're hiding the standard library's datetime module. You need to remove or rename datetime.py to something different (and if there's a datetime.pyc, just remove it).

raymondodinzeo, 2016-09-14:
Hi Dave,

After a long search I eventually found datetime.py, it is in 
/usr/lib/python3.2

Is that the problem and is there a way to fix it?

Kindest regards

Ray

waveform80, 2016-09-14:
No, that's the system's built-in datetime module; that's supposed to be there (assuming you meant /usr/lib/python3.2/datetime.py). I mean a datetime.py (or a datetime.pyc) in the same directory that you're running your script from (/home/pi). Can you paste in the result of running the following from the command line:

```
ls /home/pi
```

It'll probably look something like the following but with some more things in it:

```
Desktop         Music              Public          userland
Documents       Pictures           Scratch         Videos
Downloads       oldconffiles       Templates
```

To copy and paste here, just select the text with the mouse then middle-click to paste it into a browser window (or a text editor or whatever you like).

raymondodinzeo, 2016-09-14:
Hello Dave,

Have listed my home directory and there they were! I was looking in the 
directory of the picamera module. Sorry got the wrong end of the stick.

The script runs without errors now that I have followed the instructions 
you gave in your first email.

Thank you very much.

kindest regards

Ray

waveform80, 2016-09-14:
Ah, glad to hear it!

closed 2016-09-14 by waveform80