waveform80/picamera #327: Picamera' s real frame rate slower than speficied

Amrosik, 2016-09-14:
Hello, I followed [this blog](http://www.pyimagesearch.com/2015/12/28/increasing-raspberry-pi-fps-with-python-and-opencv/#comment-405357) to use threaded frame-grabbing. The PiVideoStream class presented there is using capture_continuous, which is updated in a seperate thread.

What I recently found with the following script, is that sometimes the update isn't done consistently with the speed necessary to maintain the fps specified:

    from __future__ import print_function
    from imutils.video.pivideostream import PiVideoStream
    from imutils.video import FPS
    import imutils
    import time
    import cv2

    # function to return the difference of elementsums between consecutive frames. If it returns 0, 
    # then  the frames are identical with a high probability

    def diff(frame1,frame2):
        return(cv2.sumElems(frame1)[0]-cv2.sumElems(frame2)[0])

    vs = PiVideoStream(resolution=(720,720),framerate=40).start()
    time.sleep(2.0)
    fps = FPS().start()
    diffs=[]
    frame2 = vs.read()
    while fps._numFrames < 50:
        frame1 = frame2
        time.sleep(0.025)
        frame2 = vs.read()
        diffs.append(diff(frame1,frame2))
        fps.update()

    fps.stop()
    print("[INFO] elasped time: {:.2f}".format(fps.elapsed()))
    print("[INFO] approx. FPS: {:.2f}".format(fps.fps()))
    print(diffs)

    cv2.destroyAllWindows()
    vs.stop()

The script captures consecutive frames and stores their difference of the elementsums in `diffs`.
Introducing a `sleep(0.025)` into the `while`-loop, the loop is able to process about 23fps. Setting the camera fps to 40, `diffs` shouldn't contain any zeros. But it does. 

I tried 2 different camera modules. Both exactly same problem. In fact, up to 40% of the list elements are zero. sometimes I have three zeros in a row.

I guess there is a problem with the implementation of threading.

One little extra:

If you modify the PiVideoStream class by adding the option of speficying the camera.sensor_mode, and then select mode number 7, which promises at least 40fps, not only dosn't help, but the diff() method takes longer to execute. While the loop was able to process about 23 fps, with sensor_mode=7 I only achieve 14 fps.

I have no idea whats going on

6by9, 2016-09-14:
How long does your diff function take to run? You haven't factored that in at all when you sleep for a full 40fps frame duration (25ms).

If you really want to measure frame rate, then each and every frame coming from the GPU has a timestamp. Is the delta between those correct for the requested frame rate?

Selecting mode 7 will limit  the sensor to producing VGA. There is no point in asking for 720x720 as the SoC then has to crop 640x480 to 480x480, and upscale that to 720x720. Image quality will suffer on any upscale.
BGR and RGB formats have a processing overhead over YUV formats. That will be amplified by having that upscale.

Amrosik, 2016-09-14:
> How long does your diff function take to run? You haven't factored that in at all when you sleep for a full 40fps frame duration (25ms).

Yes, I didn't take `diff` explicitly into account, but as I said: the loop is able to throughput 23 frames per second according to imutils.video.FPS, including the sleep duration and the duration of a `diff` execution. Setting the sleep duration to 25ms ensures, that the loop is not going to throughput more than 40fps, no matter how fast or slow `diff` is. That was the only purpose of `sleep()`.

> If you really want to measure frame rate, then each and every frame coming from the GPU has a timestamp. Is the delta between those correct for the requested frame rate?

How can I extract the timestamp?

> Selecting mode 7 will limit the sensor to producing VGA. There is no point in asking for 720x720 as the SoC then has to crop 640x480 to 480x480, and upscale that to 720x720. Image quality will suffer on any upscale.
BGR and RGB formats have a processing overhead over YUV formats. That will be amplified by having that upscale.

I didnt actually asked for 720x720 when I speficied the sensor_mode. And also, PiVideoStream is using PiRGBArray anyways, not yuv.

6by9, 2016-09-14:
To clarify, I'm one of the Pi firmware developers, so I'm advising on what the GPU is likely to be up to.
PiCamera is waveform80's handiwork, and you're using yet another library with presumably https://github.com/jrosebr1/imutils. Your problem could be in any of those layers, or your own code, so you need to identify which.

> Yes, I didn't take diff explicitly into account, but as I said: the loop is able to throughput 23 frames per second according to imutils.video.FPS, including the sleep duration and the duration of a [...]

And you're reporting that the throughput is significantly below 40fps. If ```diff``` takes 25ms, then you're not going to get 40fps, and are likely to be pushing it to get 20fps.
It sounds like what you really need is a "read and dequeue" or a blocking read from PiVideoStream so you only read each frame once - that's not a function of Picamera.

> How can I extract the timestamp?

http://picamera.readthedocs.io/en/release-1.12/api_camera.html#picamera.PiVideoFrame.timestamp

> I didnt actually asked for 720x720 when I speficied the sensor_mode. And also, PiVideoStream is using PiRGBArray anyways, not yuv.

Yes, and I'm saying that RGB has an extra overhead. There are limits to the GPU processing power, and the results of exceeding that limit is dropped frames. If you can use YUV (preferably I420) or greyscale, then you can improve performance if you're on the edge.

Amrosik, 2016-09-14:
> If diff takes 25ms, then you're not going to get 40fps, and are likely to be pushing it to get 20fps.
It sounds like what you really need is a "read and dequeue" or a blocking read from PiVideoStream so you only read each frame once - that's not a function of Picamera.

sorry if I expressed myself inacurately. you mean that the processing rate is not going to be higher than 20fps? thats true. But with "fps" I didnt mean the processing rate. If you take a look at the [source](https://github.com/jrosebr1/imutils/blob/master/imutils/video/pivideostream.py),  `frame` should be refreshed in a seperate thread (did I misunderstand this?). Setting the framerate parameter to 40 should be enough to catch a completely new frame each time read() is executed, given that its execution rate is lower than 40 per second, which is the case for my script, right?

6by9, 2016-09-14:
> sorry if I expressed myself inacurately. you mean that the processing rate is not going to be higher than 20fps? thats true. But with "fps" I didnt mean the processing rate. If you take a look at [...]

It should get a new frame, but it could also have dropped a frame before getting the new one.

The other thing you don't say is what your scene is. Is there anything moving in it? If not then couldn't diffs come out as a match even when it is a different frame?

My view is you're trying to measure fps in totally the wrong way. Detail of what you're doing in Python isn't my area of expertise - give me C and assembler any day.

Amrosik, 2016-09-15:
The scene is non-static. There is also a significant amount of noise.

I guess the main thread, where the image processing is running is somehow affecing the CPU time of the thread, which is responsible to seek(0) and truncate() the stream. So the CPU is spending less time on that thread, so it may occur that I execute read() before the truncation. 

What do you say? If this is true, then there is no point in doing a threaded stream. There is an example of threading in the picamera [docs](http://picamera.readthedocs.io/en/release-1.12/recipes2.html?highlight=rapid%20capture#rapid-capture-and-processing)

6by9, 2016-09-15:
Silly question: you are on a Pi2 or 3, I assume. If Pi1 or 0 then you're at the mercy of the scheduler as to which thread gets to run when on the single core that is available.

Python's not my language. I can give you a set of commands that should be able to stream VGA @ 90fps I420. There is a forum post somewhere that lists the results for RGB3 and BGR3, but my search-foo is failing me at the moment in finding it.
```
sudo modprobe bcm2835-v4l2
sudo apt-get install v4l-utils
v4l2-ctl -p 90
v4l2-ctl --set-fmt-video=width=640,height=480,pixelformat=I420
v4l2-ctl --stream-mmap=3 --stream-count=1000 --stream-to=/dev/null
```
That should print out the number of frames received each second as the frame rate. Replace I420 with RGB3 or BGR3 for the 24bpp RGB formats.
As long as the GPU is delivering what has been promised (and VGA 90fps was only promised as encoded video or I420), then I'm happy as the problem lives elsewhere.

6by9, 2016-09-15:
Ah, search-foo restored, but mainly as I'd misremembered the test criteria.
https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=107571&p=740712#p740800
At 5MPix:
- BGR3 - 9 fps
- RGB3 - 11-12 fps
- l420 - 15 fps
VGA is approx 1/16 of 5MPix, so a linear scale up would say we could hopefully achieve 147fps even in BGR3. It won't be linear, but that should be a big enough margin that it isn't an issue.

Amrosik, 2016-09-15:
is it true that the polling of frames from a `capture_continuous` (i.e. executing` .seek(0)` and `.truncate()`) takes at least 0.1 seconds? In PiVideoStream the polling is solved via a for loop:

```
for f in stream:
    frame=f.array
    array.seek(0)
    array.truncate()
```

Is there a faster way of polling frames out of a stream?

Amrosik, 2016-09-18:
solved via:

```

import io
import time
from picamera import PiCamera
import cv2
import numpy
from threading import Thread
import time

w=640
h=480
fw=(w+31)//32*32
fh=(h+15)//16*16
Y=None
camera=PiCamera()
camera.resolution = (w, h)
camera.framerate = 40
camera.exposure_mode='off'
camera.awb_mode='off'
camera.awb_gains=1
camera.shutter_speed=15000
stream_done=False
cv2.namedWindow("detection", cv2.WINDOW_AUTOSIZE)

def diff(frame1,frame2):
    return int(cv2.sumElems(frame1)[0]-cv2.sumElems(frame2)[0])

def outputs():
    global Y
    stream = io.BytesIO()
    while not stream_done:

        yield stream

        stream.seek(0)
        Y=numpy.fromstring(stream.getvalue(),dtype=numpy.uint8,count=fw*fh).\
           reshape((fh,fw))

        stream.seek(0)
        stream.truncate()

def capt():
    camera.capture_sequence(outputs(), 'yuv', use_video_port=True)
    pass

def startcapt():
    t = Thread(target=capt, args=())
    t.daemon = True
    t.start()

##TEST 
startcapt()
time.sleep(0.5)
diffs=[]
frame2=Y
while cv2.waitKey(1) != ord("q"):
    #cv2.imshow('detection',Y)
    frame1 = frame2
    time.sleep(0.025)
    diffs.append(diff(frame1,frame2))

stream_done=True
print(diffs)
cv2.destroyAllWindows()
camera.close()
```

waveform80, 2017-02-23:
Looks like this one got resolved; do feel free to re-open if you've any further queries regarding this script.

closed 2017-02-23 by waveform80

tinybrainrobotics, 2019-03-22:
The python script above shows 640x480 at 40fps, but I haven't been able to get 90.

tinybrainrobotics, 2019-03-22:
I'm also getting only 8fps @ 1640x1232