waveform80/picamera #329: long exposures and capture timeout

Amrosik, 2016-09-25:
The V2-Module is able to handle 10 second exposure times. I am not even able to do 6 seconds, since I constantly get this error in python:

"Timed out waiting for capture to end"

With the old firmware I was able to handle 6 seconds, because I was able to increase the PiCamera.CAPTURE_TIMEOUT attribute, which is set to 30s by default. Now, for some reason it is set read-only. Why? I sure it will solve all problems.

Amrosik, 2016-09-25:
Sloved. I dont know why I get the read-only message. but it has nothing to do with the picamera library, I guess it has something to do with root privileges. I just made that change directly to the camera.py script with sudo nano.

6by9, 2016-09-28:
Please close the issue if you've fixed your problem.

waveform80, 2016-11-14:
The read-only nature of `CAPTURE_TIMEOUT` is actually something I inadvertently changed in 1.12; it's read-write on the *class* but not on camera *instances* anymore. So, if you need to fiddle with this, replace the following:

```python
import picamera
camera = picamera.PiCamera()
camera.CAPTURE_TIMEOUT = 60
```

With this:

```python
import picameera
camera = picamera.PiCamera()
picamera.PiCamera.CAPTURE_TIMEOUT = 60
```

It'll still work the same as before in this case.

The reason behind the change is that the class now uses `__slots__` to limit the set of attributes that can be set against instances. The reason for doing *that* is that it helps teachers debug things in the class-room. One of the examples we saw during picademy was that if people misspelt an attribute name the script wouldn't do what they expected but nor would they receive an error. For example:

```python
import picamera
camera = picamera.PiCamera()
camera.exposure = 'off'    
```

Under 1.11 this wouldn't throw an error, but wouldn't work either (it'd just set a new attribute called "exposure" on the camera instance). Under 1.12 you'll get an `AttributeError` which makes things much easier for teachers trying to debug pupils scripts.

closed 2016-11-14 by waveform80