waveform80/picamera #330: iso attribute has no effect

Amrosik, 2016-09-25:
Basically adapting the recipe from the docs for low light capturing, a change of picamera.iso has no effect. No matter if put to zero or to 800, the outcome is exactly the same (neglecting noise). Is this meant to be so? Should the iso not affect the appearance?

6by9, 2016-09-28:
Refer to the answers on your forum post at https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=160933
ISO still allows exposure time to vary, so look at the settings recorded in the EXIF of a capture. Has it cranked the ISO up but exposure down to achieve the correct scene illumination? If the image is noiser, then probably.
NB Not a PiCamera bug if there is a bug.

waveform80, 2016-11-14:
Just to echo what @6by9's said: setting iso to 800 but leaving shutter speed to float usually results in similar output (the sensor gain is cranked up by the iso setting, but the camera's firmware rapidly reduces the shutter speed to maintain consistency). Setting both the iso *and* the shutter speed should result in some fairly significant differences.

If you want to play with this interactively, try the following gist (best run over SSH so you can see the preview on the Pi's display and the controls on your SSH client; set preview alpha if you don't want to do this):

https://gist.github.com/waveform80/7a0ca251d1c352b594c18ffb36db2588

waveform80, 2017-02-23:
Closing old tickets; do feel free to re-open if you've got a further query on this topic.

closed 2017-02-23 by waveform80