waveform80/picamera #331: Question : Auto White Balance (AWB) 1 frame on 2.

AlexDupont93, 2016-09-27:
I have a question, can we use the Auto White Balance function in PiCamera to analyse 1 frame on 2 ? Because I have a Multiplexer for 2 cameras in 1 RPi. So the first frame is from the 1rst camera, and the second one from the other one, the 3rd one from the 1rst camera, ect. ect.

So I need that the Auto White Balance use 1 frame on 2, so it will analyse the good camera.

Thanks !

6by9, 2016-09-28:
No.
The AWB algorithm is solely running on the GPU. It has no idea that you have attached a multiplexer and will just run as many frames through the algorithm as it can (it's a background thread as it is relatively computationally heavy, and doesn't have to run on every frame).
You could run AWB and then fix it (switch to "off"), or apply manual gains.

AlexDupont93, 2016-09-28:
Thanks for the reply ! With the cable I am able to cut of the I2C signal and switch to an other cam to apply these settings. So I think I will create my image analyse and apply some settings to a specific cam with the GPIO I2C switch. Thanks !

waveform80, 2016-11-14:
Out of curiosity: would AWB settings in the firmware survive the I2C switch? Or would it be necessary to set AWB gains again after such a switch? (I'm guessing the latter if the firmware sees it as a camera reset?)

6by9, 2016-11-14:
The firmware will run the analysis algorithm on frame N and apply the results to N+1. If you've switched cameras at that point then it will apply the calculated AWB values to the new camera as it has no idea that a switch has occurred.
If you're switching every frame and could synchronise that accurately, then you'd actually always be applying the AWB settings from the wrong camera. (In reality AWB runs on about 1 frame in 3 anyway).

I've never actually played with the muxes, but current hypothesis based on them failing with the V2 camera is that they actually send all I2C commands to all sensors and just mux the CSI bus, therefore the SoC never gets a timeout and reset on switch. It's likely to see errors and potentially drop one frame, but that should be the low.

waveform80, 2016-11-14:
Ahh, interesting stuff. Can't say I've ever played with a camera muxer either - hadn't heard about it being tried with the V2 modules though.

6by9, 2016-11-15:
I'm basing my comments on http://www.ivmech.com/magaza/en/development-modules-c-4/ivport-raspberry-pi-camera-module-multiplexer-p-90
> This version is not compatible with 8 MP Camera Module V2.

The V2 sensor reads back sensor specific data, so if you have 2 attached to the same I2C bus (and therefore on the same address) you end up with one of the responses colliding and being corrupted. The V2 sensor will then not work.

It also explains how they claim such rapid switching - all the sensors are streaming simultaneously, it's just the CSI output data that is being switched. At switch you are likely to get a corrupted frame, but the error checking is likely to result in that being thrown away. Not a particularly nice solution, but it will work.

closed 2017-02-23 by waveform80