waveform80/picamera #332: How to compare two H264 I-Frame ?

AlexDupont93, 2016-09-29:
Hi,
Is there a possibility to compare two H264 I-Frame that are stored in a Stream to see if there is motion ?

Thanks !

6by9, 2016-10-07:
Not between 2 I-frames without decoding the full frame and performing a motion search over the uncompressed images.
On a P-frame you'd need to at least perform CABAC or CAVLC decoding to the stream, find the motion vectors, and make a call based on the size of them and the residuals as to whether that construed motion.

waveform80, 2016-11-14:
You're better off [grabbing the motion vectors](http://picamera.readthedocs.io/en/release-1.12/recipes2.html#recording-motion-vector-data) to detect motion (in which case you can keep the H.264 output separately).

6by9, 2016-11-14:
IIRC You won't get motion vectors off the frame that becomes an I-frame as it is an unnecessary step in the encode process.