waveform80/picamera #333: Cannot capture full resolution RGB or YUV raw with camera v2.1

nncanepa, 2016-10-03:
I'm trying to capture a numpy RGB or YUV array at 3280x2464 and it fails. At lower resolutions it works. Here´s my code (Python 2.7):
```
h = 2464
w = 3280
camera = PiCamera()
camera.resolution = (w, h)
camera.framerate = Fraction(1, 2)
camera.shutter_speed = 250000
camera.iso = 100
camera.exposure_mode = 'off'
camera.rotation = 270
camera.start_preview()
sleep(5)
out = np.empty(h*w*3, dtype=np.uint8)
camera.capture(out, 'rgb')
camera.close()
pic = out.reshape(h, w,3)
```

It fails and I get this error:
`PiCameraMMALError: Failed to enable component: Out of resources (other than memory)`

Am I doing something wrong?

nncanepa, 2016-10-03:
Useing the PiYUVArray class I could capture at full resolution. Here's my code:
```
import time
import picamera
import picamera.array
import numpy as np
import matplotlib.pyplot as plt

%matplotlib inline

h = 2464
w = 3280

with picamera.PiCamera() as camera:
    with picamera.array.PiYUVArray(camera) as stream:
        camera.resolution = (w, h)
        camera.rotation = 270
        time.sleep(2)
        camera.capture(stream, 'yuv')
        # Show size of YUV data
        print(stream.array.shape)
        # Show size of RGB converted data
        print(stream.rgb_array.shape)
        yuv = stream.array
        rgb = stream.rgb_array
plt.imshow(yuv[:,:,0], cmap='gray')
```

With just a Warning:

```
/usr/lib/python3/dist-packages/picamera/encoders.py:545: PiCameraResolutionRounded: frame size rounded up from 3280x2464 to 3296x2464
  width, height, fwidth, fheight)))
```

closed 2016-10-21 by nncanepa

waveform80, 2016-11-14:
I don't think full resolution RGB captures worked with the V1 module either (at least, not according to [the note in the RGB capture recipe](http://picamera.readthedocs.io/en/release-1.12/recipes2.html#unencoded-image-capture-rgb-format)), but I think YUV ones did (not too surprising given RGB is by far the largest format and YUV is half its size given the quarter-res chroma data).

Incidentally, merely accessing `PiYUVArray.rgb_array` will cause a lengthy calculation of the RGB data from the YUV data (lengthy because it's on the CPU); if you don't need the RGB values (e.g. you're happy with YUV) chopping out that stuff will significantly speed up the script.