waveform80/picamera #334: "Failed to create MMAL component vc.camera_info: Function not implemented" on Yocto image

azasypkin, 2016-10-05:
There is a chance that something wrong with the Yocto image itself, but it looks very weird that `raspistill` (IIRC it also uses `libmmal` under the hood) works and `picamera` doesn't. Do you have any idea @waveform80? Thanks!

* Raspberry Pi 2 Model B;
* Raspberry Pi Camera v2.1;
* [rpi-test-image Yocto image](http://git.yoctoproject.org/cgit/cgit.cgi/meta-raspberrypi/about/?h=master) + one additional package `python-pip`;
* Camera is enabled and 256Mb is allocated for GPU (start_x=1 gpu_mem=256 in config.txt);
* Kernel version is `Linux raspberrypi2 4.4.23 #1 SMP Tue Oct 4 14:55:18 UTC 2016 armv7l armv7l armv7l GNU/Linux`;
* And output from `ldconfig`:

```
root@raspberrypi2:~# ldconfig -p | grep libmmal
	libmmal_vc_client.so (libc6) => /usr/lib/libmmal_vc_client.so
	libmmal_util.so (libc6) => /usr/lib/libmmal_util.so
	libmmal_core.so (libc6) => /usr/lib/libmmal_core.so
	libmmal_components.so (libc6) => /usr/lib/libmmal_components.so
	libmmal.so (libc6) => /usr/lib/libmmal.so
```

Everything works fine with `raspistill` - no problems whatsoever. But the following code is failing (Python 2.7, `picamera 1.12` installed via `pip install picamera[array]`):
```
import picamera
camera = picamer.PiCamera()
``` 
Here is a verbose python output:
```
Python 2.7.11 (default, Sep 29 2016, 19:22:22) 
[GCC 5.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
import picamera # directory /usr/lib/python2.7/site-packages/picamera
# /usr/lib/python2.7/site-packages/picamera/__init__.pyc matches /usr/lib/python2.7/site-packages/picamera/__init__.py
import picamera # precompiled from /usr/lib/python2.7/site-packages/picamera/__init__.pyc
# /usr/lib/python2.7/__future__.pyc matches /usr/lib/python2.7/__future__.py
import __future__ # precompiled from /usr/lib/python2.7/__future__.pyc
# /usr/lib/python2.7/site-packages/picamera/exc.pyc matches /usr/lib/python2.7/site-packages/picamera/exc.py
import picamera.exc # precompiled from /usr/lib/python2.7/site-packages/picamera/exc.pyc
# /usr/lib/python2.7/site-packages/picamera/mmal.pyc matches /usr/lib/python2.7/site-packages/picamera/mmal.py
import picamera.mmal # precompiled from /usr/lib/python2.7/site-packages/picamera/mmal.pyc
import ctypes # directory /usr/lib/python2.7/ctypes
# /usr/lib/python2.7/ctypes/__init__.pyc matches /usr/lib/python2.7/ctypes/__init__.py
import ctypes # precompiled from /usr/lib/python2.7/ctypes/__init__.pyc
dlopen("/usr/lib/python2.7/lib-dynload/_ctypes.so", 2);
import _ctypes # dynamically loaded from /usr/lib/python2.7/lib-dynload/_ctypes.so
# /usr/lib/python2.7/struct.pyc matches /usr/lib/python2.7/struct.py
import struct # precompiled from /usr/lib/python2.7/struct.pyc
dlopen("/usr/lib/python2.7/lib-dynload/_struct.so", 2);
import _struct # dynamically loaded from /usr/lib/python2.7/lib-dynload/_struct.so
# /usr/lib/python2.7/ctypes/_endian.pyc matches /usr/lib/python2.7/ctypes/_endian.py
import ctypes._endian # precompiled from /usr/lib/python2.7/ctypes/_endian.pyc
# /usr/lib/python2.7/site-packages/picamera/camera.pyc matches /usr/lib/python2.7/site-packages/picamera/camera.py
import picamera.camera # precompiled from /usr/lib/python2.7/site-packages/picamera/camera.pyc
dlopen("/usr/lib/python2.7/lib-dynload/datetime.so", 2);
import datetime # dynamically loaded from /usr/lib/python2.7/lib-dynload/datetime.so
# /usr/lib/python2.7/mimetypes.pyc matches /usr/lib/python2.7/mimetypes.py
import mimetypes # precompiled from /usr/lib/python2.7/mimetypes.pyc
# /usr/lib/python2.7/urllib.pyc matches /usr/lib/python2.7/urllib.py
import urllib # precompiled from /usr/lib/python2.7/urllib.pyc
# /usr/lib/python2.7/string.pyc matches /usr/lib/python2.7/string.py
import string # precompiled from /usr/lib/python2.7/string.pyc
dlopen("/usr/lib/python2.7/lib-dynload/strop.so", 2);
import strop # dynamically loaded from /usr/lib/python2.7/lib-dynload/strop.so
# /usr/lib/python2.7/socket.pyc matches /usr/lib/python2.7/socket.py
import socket # precompiled from /usr/lib/python2.7/socket.pyc
dlopen("/usr/lib/python2.7/lib-dynload/_socket.so", 2);
import _socket # dynamically loaded from /usr/lib/python2.7/lib-dynload/_socket.so
# /usr/lib/python2.7/functools.pyc matches /usr/lib/python2.7/functools.py
import functools # precompiled from /usr/lib/python2.7/functools.pyc
dlopen("/usr/lib/python2.7/lib-dynload/_functools.so", 2);
import _functools # dynamically loaded from /usr/lib/python2.7/lib-dynload/_functools.so
dlopen("/usr/lib/python2.7/lib-dynload/_ssl.so", 2);
import _ssl # dynamically loaded from /usr/lib/python2.7/lib-dynload/_ssl.so
[... 60 lines cut ...]
import picamera.renderers # precompiled from /usr/lib/python2.7/site-packages/picamera/renderers.pyc
# /usr/lib/python2.7/site-packages/picamera/color.pyc matches /usr/lib/python2.7/site-packages/picamera/color.py
import picamera.color # precompiled from /usr/lib/python2.7/site-packages/picamera/color.pyc
# /usr/lib/python2.7/colorsys.pyc matches /usr/lib/python2.7/colorsys.py
import colorsys # precompiled from /usr/lib/python2.7/colorsys.pyc
import RPi # directory /usr/lib/python2.7/site-packages/RPi
# /usr/lib/python2.7/site-packages/RPi/__init__.pyc matches /usr/lib/python2.7/site-packages/RPi/__init__.py
import RPi # precompiled from /usr/lib/python2.7/site-packages/RPi/__init__.pyc
dlopen("/usr/lib/python2.7/site-packages/RPi/GPIO.so", 2);
import RPi.GPIO # dynamically loaded from /usr/lib/python2.7/site-packages/RPi/GPIO.so
mmal: mmal_component_create_core: could not find component 'vc.camera_info'
Traceback (most recent call last):
  File "test.py", line 2, in <module>
    camera = picamera.PiCamera()
  File "/usr/lib/python2.7/site-packages/picamera/camera.py", line 367, in __init__
    with mo.MMALCameraInfo() as camera_info:
  File "/usr/lib/python2.7/site-packages/picamera/mmalobj.py", line 1412, in __init__
    super(MMALCameraInfo, self).__init__()
  File "/usr/lib/python2.7/site-packages/picamera/mmalobj.py", line 446, in __init__
    prefix="Failed to create MMAL component %s" % self.component_type)
  File "/usr/lib/python2.7/site-packages/picamera/exc.py", line 157, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to create MMAL component vc.camera_info: Function not implemented
....
```

6by9, 2016-10-06:
Sounds like https://github.com/raspberrypi/userland/issues/178
Have you been rebuilding userland with modified build parameters (ie  '--no-as-needed')?

azasypkin, 2016-10-07:
Hey @6by9, thanks for the hint. Currently I use `master` branch of `meta-raspberry` and there is `--no-as-needed` [1] in the `userland` recipe. With this both `raspistill` and `raspivid` work fine, but not `picamera` python lib. Out of curiosity I was trying to remove `--no-as-needed` and obviously nothing worked after that :) Or I should try to modify it in some other way?

Also wondering if it's somehow related to the fact that `picamera` is not included into yocto image from the start and I install it via `pip install picamera[array]` later on.

[1] https://git.yoctoproject.org/cgit/cgit.cgi/meta-raspberrypi/tree/recipes-graphics/userland/userland_git.bb#n41

waveform80, 2016-11-14:
I'm afraid this is all rather beyond my knowledge (I've never dealt with the `--as-needed` switch or gcc's `__attribute__` mechanism), so I doubt I'll be able to shed any light on this. The only thing I would mention is that `ctypes` in Python (which picamera uses to access libmmal) will be loading the library with `dlopen` whereas I'd assume raspistill / raspivid will be dynamically linked. I've no idea if/how that could make any difference though.

The manner of picamera's installation shouldn't make any difference; the deb on Raspbian is effectively the result of running pip install, and it certainly won't change the manner in which it accesses libmmal.

dan-santana, 2018-08-16:
It seems that picamera uses `libmmal.so` while `raspivid` and `raspistill` links against `libmmal_{core,util,vc_client}.so` directly. On Alpine Linux I added the `-DCMAKE_EXE_LINKER_FLAGS='-Wl,--no-as-needed'` to `cmake` (as in the [Yocto package](https://git.yoctoproject.org/cgit/cgit.cgi/meta-raspberrypi/tree/recipes-graphics/userland/userland_git.bb#n50)) and `raspivid`/`raspistill` worked fine but picamera gave exactly the same "Function not implemented" errors (as libmmal still wasn't properly linked to the relevant libs), so I compiled using `LDFLAGS="-Wl,--no-as-needed"` instead (as in the [Void Linux package](https://github.com/void-linux/void-packages/blob/4e73c26f56504ee1ef1884204cf53a24c0bd21e9/srcpkgs/rpi-userland/template#L18)) and it worked correctly.

`ldd /opt/vc/lib/libmmal.so` (using `-DCMAKE_EXE_LINKER_FLAGS='-Wl,--no-as-needed'`):
```
	ldd (0xb6ec1000)
	libmmal_core.so => /opt/vc/lib/libmmal_core.so (0xb6e8f000)
	libmmal_util.so => /opt/vc/lib/libmmal_util.so (0xb6e71000)
	libvcos.so => /opt/vc/lib/libvcos.so (0xb6e58000)
	libc.musl-armhf.so.1 => ldd (0xb6ec1000)
```

`ldd /opt/vc/lib/libmmal.so` (using `LDFLAGS="-Wl,--no-as-needed"`):
```
	ldd (0xb6eb6000)
	libmmal_vc_client.so => /opt/vc/lib/libmmal_vc_client.so (0xb6e87000)
	libmmal_components.so => /opt/vc/lib/libmmal_components.so (0xb6e6c000)
	libvchiq_arm.so => /opt/vc/lib/libvchiq_arm.so (0xb6e56000)
	libvcsm.so => /opt/vc/lib/libvcsm.so (0xb6e40000)
	libmmal_core.so => /opt/vc/lib/libmmal_core.so (0xb6e22000)
	libmmal_util.so => /opt/vc/lib/libmmal_util.so (0xb6e04000)
	libcontainers.so => /opt/vc/lib/libcontainers.so (0xb6de1000)
	libvcos.so => /opt/vc/lib/libvcos.so (0xb6dc8000)
	libc.musl-armhf.so.1 => ldd (0xb6eb6000)
```

It certainly isn't a bug in `picamera`, so it seems the only workaround that could be done here would be to use the `libmmal_*.so` directly as `raspivid`/`raspistill` does (idk if that's feasible/possible though).

dan-santana, 2018-08-16:
NOTE: applying https://github.com/raspberrypi/userland/pull/459 locally still gave me the same results as using `-DCMAKE_EXE_LINKER_FLAGS` i.e. no libs linked to `libmmal.so`, so maybe this is relevant to you @6by9  :)

6by9, 2018-08-16:
The linker flags are needed when compiling picamera, not libmmal.

https://github.com/raspberrypi/userland/pull/459 only changes the dependencies of the standard apps, not the libraries.

libmmal does NOT depend on  libmmal_vc_client. If you wish to use the VPU components then your application needs to link against it, and to NOT over optimise and end up removing the constructor.
https://github.com/raspberrypi/userland/issues/178 / https://github.com/raspberrypi/userland/issues/303#issuecomment-204394592

waveform80, 2018-09-29:
That's ... odd; picamera isn't compiled (in any meaningful sense). It's just dynamically loading what it needs from the C libs identified in picamera/mmal.py. I'm not entirely familiar with the guts of CPython's ctypes module but I'd guess ultimately it's just calling dlopen and dlsym to grab what it needs from libmmal and libbcm_host. I'm afraid I can't shed much more light on this!

eswarrajan-subrmanian, 2020-04-02:
I am also facing a similar error while building userland.
I am using the attached python code for livestreaming.

All working fine till this morning, until I built userland for cec-client.
Later I am facing this issue everytime. Could someone share me the libmmal_vc_client.so with vc.camera_info included.

Error:
pimate@pimate-eswar:~$ python /home/pimate/ESWAR/LIVE/live.py
mmal: mmal_component_create_core: could not find component 'vc.camera_info'
Traceback (most recent call last):
  File "/home/pimate/ESWAR/LIVE/live.py", line 8, in <module>
    camera = PiCamera()
  File "/usr/local/lib/python3.6/dist-packages/picamera/camera.py", line 367, in __init__
    with mo.MMALCameraInfo() as camera_info:
  File "/usr/local/lib/python3.6/dist-packages/picamera/mmalobj.py", line 2346, in __init__
    super(MMALCameraInfo, self).__init__()
  File "/usr/local/lib/python3.6/dist-packages/picamera/mmalobj.py", line 633, in __init__
    prefix="Failed to create MMAL component %s" % self.component_type)
  File "/usr/local/lib/python3.6/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to create MMAL component b'vc.camera_info': Function not implemented
pimate@pimate-eswar:~$

[live.txt](https://github.com/waveform80/picamera/files/4422392/live.txt)

eswarrajan-subrmanian, 2020-04-02:
> 
> 
> I am also facing a similar error while building userland.
> I am using the attached python code for livestreaming.
> 
> All working fine till this morning, until I built userland for cec-client.
> Later I am facing this issue everytime. Could someone share me the libmmal_vc_client.so with vc.camera_info included.
> 
> Error:
> python /home/pimate/ESWAR/LIVE/live.py
> mmal: mmal_component_create_core: could not find component 'vc.camera_info'
> Traceback (most recent call last):
> File "/home/pimate/ESWAR/LIVE/live.py", line 8, in
> camera = PiCamera()
> File "/usr/local/lib/python3.6/dist-packages/picamera/camera.py", line 367, in **init**
> with mo.MMALCameraInfo() as camera_info:
> File "/usr/local/lib/python3.6/dist-packages/picamera/mmalobj.py", line 2346, in **init**
> super(MMALCameraInfo, self).**init**()
> File "/usr/local/lib/python3.6/dist-packages/picamera/mmalobj.py", line 633, in **init**
> prefix="Failed to create MMAL component %s" % self.component_type)
> File "/usr/local/lib/python3.6/dist-packages/picamera/exc.py", line 184, in mmal_check
> raise PiCameraMMALError(status, prefix)
> picamera.exc.PiCameraMMALError: Failed to create MMAL component b'vc.camera_info': Function not implemented
> 
> [live.txt](https://github.com/waveform80/picamera/files/4422392/live.txt)

Where do i have to put the following to avoid this error?
SET( CMAKE_EXE_LINKER_FLAGS "-Wl,--no-as-needed" )

eswarrajan-subrmanian, 2020-04-03:
I tried the following and the issue is resolved.

1. sudo rm -rf /opt/vc
2. sudo apt-get update
3. sudo apt-get install --reinstall libraspberrypi0 libraspberrypi-dev libraspberrypi-doc libraspberrypi-bin

mrusme, 2020-09-10:
I'm also experiencing this issue with a self-compiled version of the raspberry userland repo on the latest Ubuntu Server image for the Raspberry Pi 4. I've added `SET( CMAKE_EXE_LINKER_FLAGS "-Wl,--no-as-needed" )` to the `CMakeLists.txt` of that repo and ran the `buildme` script. However, picamera says:

```
mmal: mmal_component_create_core: could not find component 'vc.camera_info'
Traceback (most recent call last):
  File "picam.py", line 171, in <module>
    main()
  File "picam.py", line 143, in main
    with picamera.PiCamera(
  File "/usr/local/lib/python3.8/dist-packages/picamera/camera.py", line 367, in __init__
    with mo.MMALCameraInfo() as camera_info:
  File "/usr/local/lib/python3.8/dist-packages/picamera/mmalobj.py", line 2346, in __init__
    super(MMALCameraInfo, self).__init__()
  File "/usr/local/lib/python3.8/dist-packages/picamera/mmalobj.py", line 631, in __init__
    mmal_check(
  File "/usr/local/lib/python3.8/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to create MMAL component b'vc.camera_info': Function not implemented
```

Does anyone have another hint here? Thanks!

caiostringari, 2021-04-20:
Not a very good fix, but copying libmmal.so from a working version of Raspbian (/opt/vc/lib/libmmal.so) to /usr/lib/arm-linux-gnueabihf/libmmal.so on Ubuntu Mate 20.04 worked for me.