waveform80/picamera #335: Saving video as Motion JPEG (MJPEG) file

portsample, 2016-10-09:
I would like to be able to use Picamera to record and store video in the motion JPEG (MJPEG) format. I've been using variations on the following code,

#!/usr/bin/env python
print "Here we go!"
import picamera
import time
camera = picamera.PiCamera()
camera.resolution = (640, 480)
camera.framerate = 10
camera.start_preview()
time.sleep(5)
camera.start_recording('salmonCam.mjpeg')
camera.wait_recording(60)
camera.stop_recording()
print "All Done!"

The file produced from the above script does not play in VLC. Varying framerate and camera.wait_recording() interval produces files generally of anticipated size. However, when played with VLC only a  single JPEG image is shown. I have succeeded in writing to a flash USB drive in addition to writing to the sdcard. Am I missing something with my implementation? Note: device is RP3 model B, camera module is v2, card is SanDisk Ultra 128GB microSDXC UHS-I (class 10). I am using Raspbian release 1.0 v9/19. Attached is a produced MJPEG file. Thanks.
[salmonCam-168.mjpeg.zip](https://github.com/waveform80/picamera/files/518015/salmonCam-168.mjpeg.zip)

Thanks.

6by9, 2016-10-09:
It's a limitation in VLC, not in PiCamera or the Pi firmware.
VLC can not handle MJPEG elementary streams (ie outside a container such as MP4, MKV, AVI, etc). It seemingly detects the data as a single JPEG and displays that.
Use something like ```avconv -i salmonCam-168.mjpeg -vcodec copy muxed.mkv``` to multiplex it into a container, and VLC will play that quite happily.

The version of Mplayer installed on my Ubuntu 14.04 machine seems to play it quite happily even without muxing the stream. That may be an alternative for you.

portsample, 2016-10-09:
I'm only slightly confused. I am able top play MJPEG files in VLC that are generated in my Android application. The MJPEG file is produced using Shawn VanEvery's code that I've implemented in my application (https://github.com/portsample/salmonCam). Attached is an MJPEG I just recorded w/salmonCam that plays on VLC. 

[salmonCam-20161009130026.mjpeg.zip](https://github.com/waveform80/picamera/files/518556/salmonCam-20161009130026.mjpeg.zip)

I was wondering if the header frames between individual JPEGs in the Picamera  MJPEGs were somehow not of the standard used by VLC. Thanks for taking the time to respond and addressing this question.

6by9, 2016-10-10:
Please find a copy of the MJPEG standard to analyse the streams against. That'll be tricky as there isn't one single document (https://en.wikipedia.org/wiki/Motion_JPEG#Disadvantages).

Just looking at your two files, the one under Android has a MIME like header
```

--myboundary
Content Type: image/jpeg
Image Size: 9729 bytes
AbsTime: 20161009130026
JpegQual: 50%
FPS: 4
CamcorderFrame: 480 x 720
CamcorderBitrate: 6000000
Light level: 4.0
```
before the FFD8 JPEG start marker of every frame. That appears to be the difference. If I prepend the first frame in the Pi file with that then VLC 2.1.3 is happy (some frames are then corrupted, but that is probably because VLC is then looking for further similar markers, and they aren't there).

To my mind that is effectively a very basic form of muxing, or at least signalling to VLC the correct format. That is not the job of the stream, but of the container. We already know that VLC does the correct thing given the stream in a valid container.
I can gather views from the others developing the firmware, but I'm not convinced that this is a worthwhile change, particularly as MPlayer seems to do the correct thing with the current streams.
 from the others developing the firmware, but I'm not convinced that this is a worthwhile change, particularly as MPlayer seems to do the correct thing with the current streams.

NB This should really have been reported in https://github.com/raspberrypi/firmware/issues. PiCamera is only a wrapper over the streams the firmware produces. Don't bother raising an issue there now though as we have the root cause.

6by9, 2016-10-10:
I've had an email exchange with the other developer who knows codecs. He also feels that the codec is doing nothing wrong as these headers have all the hallmarks of being a container, and the encoder produces a raw stream.
I'll compare it tonight to what my Logitech webcam produces as an MJPEG stream as a reference. I suspect that too will produce a raw stream without headers, but I'm prepared to be wrong.

One option that will be investigated is adding an option to raspivid (and PiCamera could copy) for the app to insert the header as an option. At least that way the files will play back in VLC.

6by9, 2016-10-10:
I remembered that my Ubuntu laptop has a webcam too. It also produces just the raw stream via V4L2, so VLC again refuses to play it. Pi is therefore not the only device that VLC has a problem with.

This is really a question over supporting containers. We've resisted adding that to raspivid for a long time as it adds a BIG dependency. Perhaps it needs to be revisited...
What @waveform80 chooses to do for PiCamera is totally up to him. Again adding container support is unlikely to be trivial, and can take you away from the underlying hardware and systems.

portsample, 2016-10-10:
Thank you again for responding. VLC is available for a wide variety of operating systems. It is the standard viewer for MJPEG files in many technical environments. I added the header comments that you refer to as a method of adding metadata to each of the captured images. I hope RP developers will see the value in assigning a format to the MJPEG files created that is standard with as many of the widely used viewers as possible.

6by9, 2016-10-10:
I fully appreciate that VLC is a standard viewer, however the issue still appears to be that VLC is unable to accept an MJPEG elementary stream (to the extent that such a thing is defined). It incorrectly determines that it is a single JPEG.
The stream produced by the Pi is fully compliant as an elementary stream in that it is a sequence of concatenated JPEG frames.

Capture a raw stream from any V4L2 webcam that supports MJPEG using
```v4l2-ctl -v width=640,height=480,pixelformat=MJPG
v4l2-ctl --stream-mmap=3 --stream-count=100 --stream-to=file.mjpeg```
and off my laptop VLC is unable to play the file satisfactorily. I do see it detect the stream as MJPEG (MJPG) instead J420, but not usable.
The only difference I can see on analysing the stream headers are that the Pi inserts a FFE0 JFIF block (mandatory in the JPEG spec), whilst the webcam doesn't. The webcam also inserts restart markers.

Mplayer and ffmpeg are both happy with the Pi stream and that produced from my webcam.

portsample, 2016-10-11:
I've noticed that inserting a space before the first JPEG marker (FFD8) allows the MJPEG file to play in VLC just fine, as well as the Mplayer viewer. 
Is this a modification that developers might consider supporting? 
Thank you for your time and patience in considering this request.

6by9, 2016-10-12:
Adding a space at the start isn't a viable solution at the codec level. Under which conditions do you add it? What happens if the client splits the stream at a random frame boundary? Then the second part has no space and so will fail to decode but the first half is OK. 

I will try to deduce why VLC gets upset. As I said, the stream from my USB webcam seems to make it slightly upset, but it still sees it as MJPEG instead of J420. 
This is still a containers issue. Adding your space is just changing the behaviour of the container/codec detection code, and in that regard is almost a container in itself.

6by9, 2016-10-12:
I've raised https://github.com/raspberrypi/userland/issues/345 to investigate possibilities.
As said earlier, this is not a PiCamera library bug, so this probably ought to be closed due to the inappropriate filing.

6by9, 2016-10-12:
```vlc --demux=mjpeg --mjpeg-fps=10 salmon-168.mjpeg ```
plays the file back as expected by forcing the demuxer and providing a frame rate. It also works if you specify the demuxer as ```avformat```.

Using demux=mjpeg doesn't work with your Android app stream, but using avformat does. Basically you need to override the VLC choice of demuxer.

waveform80, 2016-11-14:
Just to echo what @6by9's already said: the issue here is that MJPEG is a rather ill-defined format (or formats given there appear to be several), and VLC (whilst certainly very popular) is not the definer or arbiter of that format.

I'm happy to enhance picamera to produce more "friendly" MJPEG files but in this case I won't be doing anything (yet), because:

* As has been mentioned the space hack isn't a viable solution (in light of split streams); I'll add it may also fix things for VLC whilst breaking other players
* Looking at the upstream ticket (raspberrypi/userland#345) this really looks like a VLC issue more than anything (VLC seems to assume anything with a JFIF header can't be MJPEG)
* Nonetheless, it looks like there might be some possibility of the firmware being tweaked to produce some MJPEG variant which VLC would accept without adjustment (and I'm all for doing nothing when I can ;)

For now I'll keep an eye on the upstream ticket. But I'll mark this for the next version and if there's no other fix by the time I get around to releasing that, I'll add some bits to the docs about the `--demux=mjpeg` workaround because that's worth mentioning.

commit 1c1a9473 (2017-01-04): Fix #335

Add a note about --demux=jpeg
commit 569463f8 (2017-01-28): Refactor mmalobj

* Add callback capability on both pure MMAL and python-based connections
* Use a common format negotiation algorithm on both MMAL and
  python-based connections
* Permit connections to be constructed port->port rather than just
  component->port
* Fix #335; unfortunately after much experimentation the "best" fix
  means making MMAL_FOURCC's straight ctypes ints (this also fixes
  several issues with comparisons and making sets of encodings)
* Fix up the docs and mmalobj examples
closed 2017-01-04 by waveform80