waveform80/picamera #336: Cannot take long exposure (low framerate) with camera v2.1 at full resolution

nncanepa, 2016-10-11:
I cannot set the framerate below 16 when the camera resolution is at 3280x2464, I'm using a RPi 3 and the v2.1 of the camera.
I'm using the following code (from the picamera docs recipies):

``` 
import time
import picamera
import picamera.array
import numpy as np
import matplotlib.pyplot as plt
import scipy.misc

h = 2464
w = 3280
sp = 10000
iso = 100

with picamera.PiCamera() as camera:
    with picamera.array.PiYUVArray(camera) as stream:
        camera.resolution = (w, h)
        camera.iso = iso
        time.sleep(5)
        #camera.framerate = 24
        camera.shutter_speed = sp
        camera.rotation = 270
        time.sleep(5)
        camera.exposure_mode = 'off'
        camera.capture(stream, 'yuv')
        yuv = stream.array
        rgb = stream.rgb_array
```

If I use a lower resolution there's no problem in using a framerate of 1 or even Fraction(1, 6)

nncanepa, 2016-10-17:
I answered myself, giving a better read to the documentation.
I had to increment the gpu_mem in /boot/config.txt from 128 to 256.

Sorry for the noise

closed 2016-10-21 by nncanepa

waveform80, 2016-11-14:
No prob - glad you got it sorted :)