waveform80/picamera #338: Overlay Text with OpenCV in CustomEncoder

Altair68, 2016-10-13:
Is it possible to overlay a given Text with a CustomEncoder?
This is a snippet of what I'm trying to do:

```
import picamera
import picamera.mmal as mmal
import sys
import numpy as np

class MyEncoder(picamera.PiCookedVideoEncoder):
    def start(self, output, motion_output=None):
        super(MyEncoder, self).start(output, motion_output)

    def _callback_write(self, buf):

        bufData = buf[0].data

        data = np.fromstring(bufData, dtype=np.uint8)
        image = cv2.imdecode(data, 1)

        super(MyEncoder, self)._callback_write(buf)

class MyCamera(picamera.PiCamera):
    def __init__(self):
        super(MyCamera, self).__init__();

    def _get_video_encoder(self, camera_port, output_port, format, resize, **options):
        return MyEncoder(self, camera_port, output_port, format, resize, **options)

with MyCamera() as camera:
    while True:
        camera.start_recording(sys.stdout, format='h264')
        camera.wait_recording(60 *60 * 24)
        camera.stop_recording()
```

With that code "camera.stop_recording()" raises "TypeError: 'MMALBuffer' object does not support indexing".

Or is there an easier way of doing what I'm trying to do here?

Berryfier, 2016-10-14:
See my issue #337 on removing the indexing error.

Altair68, 2016-10-14:
So basically adding `key=0` as a parameter should do the trick?

Berryfier, 2016-10-16:
The interactive help identifies key=0, but this may be a default. For me the removal of [0] after buf was sufficient.

Altair68, 2016-10-16:
Ahhh now I see! 👍 Thanks a lot :)

Altair68, 2016-10-22:
Alright now that accessing the data works I need to convert the data into an opencv object which isn't working because "cv2.imdecode(data, 1)" returns a NoneType. Any ideas on that?

closed 2016-10-22 by Altair68

waveform80, 2016-11-14:
If you just want to overlay text, see [the `annotate_text` property](http://picamera.readthedocs.io/en/release-1.12/api_camera.html#picamera.PiCamera.annotate_text).

Unfortunately what you're trying to do (modify the image in the encoder class) isn't going to work: the encoder callback is at the very end of the pipeline after the image has been encoded and is about to be written to the output. What I think you're wanting to do is modify the image *before* it gets to the encoder; that's not (currently, easily) possible with picamera.