waveform80/picamera #339: Question: Can I define the ROI for 1920x1080 partial FOV?

paevenson, 2016-10-14:
Hi,

Is it possible to define the RoI on the sensor using the 1920x1080 sensor mode? I basically want to grab the left most area of the sensor while using the 1920x1080 partial FoV.

[image]

6by9, 2016-10-15:
Sorry, no. The readout area is defined in the GPU driver for the sensor.

paulhothersall, 2016-10-17:
well if you could cope with 15fps, the answer is yes*

*for a convoluted way of getting there, and may involve slight quality loss

6by9, 2016-10-17:
> well if you could cope with 15fps, the answer is yes*
*for a convoluted way of getting there, and may involve slight quality loss

I guess so, although not quite what was originally asked.
Force sensor mode 2, then specify a zoom of (x,y, 0.74, 0.5556), and as long as x < 0.26, and y < 0.44, you can move the region around. No loss in quality, just in frame rate.

paevenson, 2016-10-17:
Thanks for the responses. Yeah 15 fps is a little too low for me. I ended up with 1296x972(mode 4), defining my RoI with the zoom feature, and up scaling to 1920x1080. This gives me an FPS of 42 which is acceptable, but the aliasing from the binning/up scaling is problematic... 

I may attempt to bringing over the openGL libraries to perform some anti-aliasing... anyone else have a recommendation?

paulhothersall, 2016-10-17:
paevenson, that works probably the best for your v1 camera. try 30fps if you can, the camera tweaking seems well dialled in at this FPS more than others, for whatever reason despite the forced sensor setting should make no difference. maybe less time for the ISP fixing up the input?

I have been looking at a few other items on this toopic, so 6by9 can you comment on how to pan/scan to move the region around the WHOLE sensor area without loss of quality, BUT for a different aspect ratio?

As I understand it, you have to set the same aspect ratio for the initial res to have the full sensor , then pick your zoom, then pan, then resize. Otherwise the res set is a crop, and then you have less room to move around in.

example here, for 4:3 full frame sensor, but using a FOV of 3:4

`
camera.resolution = (1640, 1232) #for camera module v2 only

x = 0.0
y = 0.0
w = 0.439024390        # values for setting a rotated and smaller ratio
h = 0.779220779          # in this call 3:4 format, nominal 720x960 res

camera.zoom = (x, y, w, h)

camera.start_recording('zoomtest.h264', resize=(720, 960)) 

<bunch of stuff to move the ROI around, moving in multiples of 

        x by 0.000609756
        y by 0.000811688             to get single pixel increment moves  > `

6by9, do I really need to use the resize function as shown in example above, which I assume comes with some overhead to get the best result?

6by9, 2016-10-17:
I think your issue is that the mode choosing algorithm isn't making the ideal choice if you're going for weird aspect ratios. 
I'm assuming you're still wanting square pixels (ie not actually distorting stuff). I'd hope that forcing the sensor mode, setting your weird zoom, and then asking for 3:4 aspect ratio output should have worked.

IIRC the intent was that you were specifying the region that should be considered as active from the input image, and then that would be scaled/cropped to fit your output resolution. The underlying parameter is called MMAL_PARAMETER_INPUT_CROP, and the IL docs say:
```
 /* OMX_IndexConfigInputCropPercentage: Specify input crop as a percentage */
typedef struct OMX_CONFIG_INPUTCROPTYPE {
   OMX_U32 nSize;
   OMX_VERSIONTYPE nVersion;
   OMX_U32 nPortIndex;
   OMX_U32 xLeft;     /**< Fraction of the width for the top left corner of the rectangle */
   OMX_U32 xTop;      /**< Fraction of the height for the top left corner of the rectangle */
   OMX_U32 xWidth;    /**< Fraction of the image width desired */
   OMX_U32 xHeight;   /**< Fraction of the image height desired */
} OMX_CONFIG_INPUTCROPTYPE;
/*
This parameter allows the input cropping to be specified as a
percentage of the current width/height.  Required for the camera
component where the output resolution varies dependent on the port.
All percentage values are as 16p16 fixed point numbers (0x10000 =
100\%)
*/
```
Yes resize has quite a performance hit - avoid if you can.