waveform80/picamera #340: libmmal.so dependency not checked by installer

polycopter, 2016-10-20:
Installed /usr/lib/python2.7/site-packages/picamera-1.12-py2.7.egg
Processing dependencies for picamera==1.12
Finished processing dependencies for picamera==1.12
[prompt]$ python
Python 2.7.5 (default, Nov 20 2015, 02:00:19) 
[GCC 4.8.5 20150623 (Red Hat 4.8.5-4)] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import picamera
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "picamera/__init__.py", line 72, in <module>
    from picamera.exc import (
  File "picamera/exc.py", line 41, in <module>
    import picamera.mmal as mmal
  File "picamera/mmal.py", line 47, in <module>
    _lib = ct.CDLL('libmmal.so')
  File "/usr/lib64/python2.7/ctypes/__init__.py", line 360, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libmmal.so: cannot open shared object file: No such file or directory

stuaxo, 2016-11-03:
I hit the same thing in installing with pip in a virtualenv on python3.5

waveform80, 2016-11-04:
There's no (simple) way that I'm aware of to enforce / check that dependency with Python's packaging system. The deb package, however, *does* enforce that dependency by depending on the `libraspberrypi0` package which is what includes `libmmal.so`. I could add a note to the pip installation instructions, but I think that's about all I can do (I've no idea what package includes `libmmal.so`, and the associated libraries, on distros other than Raspbian).

stuaxo, 2016-11-04:
Makes sense, I was trying to run this on a PC which didn't work obviously :)

waveform80, 2016-11-04:
On a PC? Err, yeah, never gonna work there ;)

stuaxo, 2016-11-04:
I was using pi3d (which does work there) + sort of stopped thinking when I went to install requirements for some of the examples :)

polycopter, 2016-11-13:
I also was trying to install on a non-Pi device (PC, I think); I never
realized this was RPi-specific. Sorry!

waveform80, 2016-11-14:
Okay, there's evidently enough people trying to install this on PCs (?!) that I could make things a little more obvious for them. I'll throw some stuff in `setup.py` to throw an error if install is being attempted on something that's blatantly not a Pi (however, I'll deliberately only do this for the "install" command - there's good reasons for development installations on non-Pi platforms, e.g. building docs, packages, etc.)

commit b5e834fe (2016-11-14): Fix #340

Throw an execption if installation is attempted on a non-Pi platform;
deliberately avoid breaking "develop" and "bdist_egg" in these cases
otherwise we'll break things like doc-only builds, packaging, etc.
closed 2016-11-14 by waveform80