waveform80/picamera #341: Custom Output: How to convert the bytestring to a proper OpenCv Object

Altair68, 2016-10-24:
How do I convert the given byte-string inside the custom output to a numpy-array that's working with OpenCv? And how do I convert it back after altering the image?

```
from __future__ import print_function

import picamera

class MyOutput(object):
    def __init__(self):
        self.size = 0

    def write(self, s):
        self.size += len(s)
        #-->How to convert s to an OpenCv object?

    def flush(self):
        print('%d bytes would have been written' % self.size)

with picamera.PiCamera() as camera:
    camera.resolution = (640, 480)
    camera.framerate = 60
    camera.start_recording(MyOutput(), format='h264')
    camera.wait_recording(10)
    camera.stop_recording()
```

waveform80, 2016-11-14:
Guessing this is a follow-on from #338...

The output's `write` method is being called with H.264 encoded data and I get the impression you're wanting to modify the unencoded frame (using OpenCV) *before* it gets to the H.264 encoder. Unfortunately, that's not easily possible with picamera, which is hooking the camera's video output (almost) straight to the H.264 encoder in the GPU (i.e. the ARM never sees the unencoded frames).

My understanding is that it is possible to construct an MMAL pipeline (or rather pipelines) which permit the ARM to modify the unencoded frames before they reach the encoder (essentially you setup an unencoded recording, then setup a standalone H.264 encoder, grab buffers from the recording, modify them and pass them onto the encoder, writing its output wherever you want). However, I've not tried this myself hence it's not documented anywhere at the moment, and I've no idea what sort of framerate one could achieve performing the modification in Python (would probably depend on the complexity of the modification).

It's something I'd like to cover at some point, but I've no idea when that'll be at the moment.

Altair68, 2016-11-14:
Alrighty. Thanks anyways for the answer!

closed 2016-11-14 by Altair68