waveform80/picamera #342: Having difficulty with stereo captures using video ports

ggoodchild, 2016-10-30:
The following code works as expected provided the "use_video_port=True" parameter is omitted.  If it is included as written, then the output file contains the image from one camera and a field of green where the second camera's image should be.  What am I doing wrong?

`import time
import picamera

count = 0
duration = 3
fn = "seq"

cam = picamera.PiCamera(camera_num=0, stereo_mode='side-by-side', stereo_decimate=True, resolution=(1280,480), framerate=30)
cam.start_preview()
time.sleep(2)
start = time.time()

while time.time() - start < duration:
    fname = fn + str(count) + '.jpg'
    cam.capture(fname, 'jpeg', use_video_port=True)
    count = count + 1

`

6by9, 2016-10-31:
What firmware revision are you on? (```vcgencmd version```)
Green is all 0 in YUV colourspace, so something's not writing the data correctly.
I had a vague recollection of a fix having gone in for the image copy function not handling multiple channels correctly, but can't see it now.
It'll be a firmware bug rather than PiCamera itself - I'll add it to the list of things to have a look at.

ggoodchild, 2016-10-31:
Thank you for responding.
pi@raspberrypi:~ $ vcgencmd version
May 20 2016 19:05:21
Copyright (c) 2012 Broadcom
version faf071dd4885c5ac1a89483d35a5326e7f69495f (clean) (release)

6by9, 2016-10-31:
Thanks. Recent enough even if not absolute latest.

@waveform80 Could you confirm for me that with use_video_port=True, camera->output[1] goes to video_splitter->input[0] as OPAQUE, and then video_splitter->output[N] goes to image_encode->input[0] as I420?
I think the issue is that video_splitter isn't handling the multiple image channels correctly (I only fixed up camera, image_encode, video_encode, and video_render for multichannel stuff. Resize is likely to break too).

6by9, 2016-11-16:
Ping @waveform80 again.
Could you confirm for me that with use_video_port=True, camera->output[1] goes to video_splitter->input[0] as OPAQUE, and then video_splitter->output[N] goes to image_encode->input[0] as I420? It would save me a bit of time in creating a test case (I really don't do Python, and I can't remember the state of my CM anyway)

waveform80, 2016-11-16:
Sorry, completely forgot about this! Erm, good question - I can't remember exactly what happens as of 1.12 (because it tries to use OPAQUE wherever it possibly can now). I'll have a look what it does this evening.

waveform80, 2016-11-16:
Yup, it is indeed "camera->output[1] goes to video_splitter->input[0] as OPAQUE, and then video_splitter->output[N] goes to image_encode->input[0] as I420". Incidentally, for anyone that wants to debug this sort of thing I added a little something in 1.11 or thereabouts for debugging this stuff. Just set `PiEncoder.DEBUG` to `True` and each time an encoder gets started it'll dump a description of the pipeline from the encoder's output back to the camera:

```pycon
>>> import picamera
>>> camera = picamera.PiCamera()
>>> picamera.PiEncoder.DEBUG = True
>>> camera.capture('foo.jpg', use_video_port=True)
 vc.ril.camera [1]                                [0] vc.ril.video_splitter [0]                               [0] vc.ril.image_encode [0]       
   encoding    OPQV-dual         -->      OPQV-single       encoding        I420             -->             I420      encoding       JPEG      
      buf      10x128                          10x128          buf          1x3133440                   1x3133440         buf         1x81920   
     frame     1920x1088@30fps         1920x1088@0fps         frame         1920x1088@0fps         1920x1088@0fps        frame        0x0@0fps  

```

However, I have just discovered the debug stuff's broken under Py3 (doh!) so I'll get that fixed up :)

6by9, 2016-11-16:
Thank you. I'll look to fix up video_splitter then and hopefully that'll resolve this issue.

6by9, 2017-02-03:
I have firmware change that fix this, I just need to get them merged and released into a firmware image (I'd guess early next week)

ggoodchild, 2017-02-03:
Thank you very much for all your hard work.  I greatly appreciate it.

6by9, 2017-02-08:
Fix released if you do ```sudo rpi-update```.
Don't do this on a critical machine - that is the cutting edge branch so may have issues.

closed 2017-02-23 by waveform80

farzanehnakhaee70, 2017-07-30:
Unfortunately my problem has not been solved using rpi-update. what can I do?