waveform80/picamera #343: Picamera Bitrate CBR

StormPi, 2016-11-01:
Hello,
Excellent project Picamera.
I need "Picamera" to use Both video and photo.
And I need a constant bitrate or I notice that the night bitrate collapses.
For example the day, the bitrate is stable at 700 000, and at night it comes down to 26,000.
Is there a solution for the bitrate is stable at 700 000 24h/24 as raspivid?

Thank

waveform80, 2016-11-14:
What's the option you're using on raspivid to maintain a constant bitrate? As far as I'm aware, picamera configures the video encoders in a pretty similar fashion to raspivid.

StormPi, 2016-12-22:
Thank for your reply.
"With Raspivid, no option used.
For PiCamera, I noticed this with ffmepg when I want to send the stream to Youtube Live
In full light, no problem, the bitrate is found at 700 000 as against when the camera is placed in the dark, the bitrate falls to 200 000 instead of 700 000 and if put back in broad daylight, It remains at 200,000.
Similarly, if you start in the dark, the bitrate does not exceed 26,000."

StormPi, 2016-12-22:
As for me, I just tested Raspivid + ffmpeg, the bitrate also falls in the dark.
Thank

waveform80, 2016-12-22:
I tried both picamera and raspivid in a room at night, switching the light on and off while constantly monitoring the output of both. In each case switching the light off caused the output bitrate to drop to ~1Mbps (I was recording at 720p30 in both cases), and switching the light on again caused the output bitrate to rise to ~6Mbps. In other words, I can't replicate raspivid producing a constant bitrate output.

If you're using ffmpeg to post-process the output (in either case) then you can certainly get it to produce CBR output, but I'm not (currently) aware of an MMAL option to permit CBR output (MMAL is the basis upon which both raspivid and picamera are built).

6by9, 2016-12-23:
What are you trying to achieve with your constant bitrate? Is there any useful information left in the night time scene?
The standard rate control does have limits so that it doesn't end up encoding noise with more and more accuracy on otherwise flat scenes. If the image is too dark, then drop the frame rate to allow the camera to use longer exposures and pick up something useful to encode.

There is a MMAL option for nominal CBR - MMAL_VIDEO_ENCODER_RC_MODEL_CBR (https://github.com/raspberrypi/userland/blob/master/interface/mmal/mmal_parameters_video.h#L346), set via MMAL_PARAMETER_RATECONTROL. This is not exposed via Raspivid, and you also have to drop the use of CABAC encoding of the stream due to the way the codec pipeline works (means H264 base profile).
Even so this is a single pass encode, so there will always be variation in the bitrate as it will make a guess at a suitable quantisation factor at the start of a frame based on recent history, and adjusts that for the next frame. (ffmpeg is generally a two-pass encode, so analyses the frame first, determines a suitable QP, and then actually encodes with that.)

StormPi, 2016-12-23:
Thanks for your help and test.
The constant bitrate is for a weather cam. So inevitably at night there is not much going on.
But Youtube Live wants CBR: https://support.google.com/youtube/answer/2853702?hl=en
For example, from 700,000 to 30,000 at night, youtube no longer receives enough throughput and cuts the live stream.

6by9, 2016-12-24:
Do the images actually go totally black at night? If so then you are going to struggle to get the bitrate up. Drop the frame rate so that the sensor can actually capture a sensible scene, and then you should maintain a reasonable bitrate.
I doubt even with CBR requested that it will crank the QP much further (or even if it can), and it's not going to create bits just to pad the stream out.