waveform80/picamera #345: Picamera-issue with coding

richinixon, 2016-11-08:
Hello,

I am using Rpi 3 model and pi camera as a hardware part. I have installed the picamera package and i try to run the following program, i am ending up with an eror.

`import time
import picamera
with picamera.PiCamera() as camera:
    camera.start_preview()
    time.sleep(2)
    camera.capture('foo.jpg')
    camera.stop_preview()
`
Error: 
Traceback (most recent call last):
File "/home/pi/camera.py", line 4, in <module>
with picamera.Picamera() as camera:
AttributeError: 'module' object has no attribute 'Picamera'

Please do suggest on resolving the above error.

Regards,
Richi

waveform80, 2016-11-14:
Python is case sensitive; you're using `Picamera` as the class-name in your script rather than `PiCamera` (note the capital C in camera).

richinixon, 2016-11-15:
Thank you.. issue resolved :)

closed 2016-11-16 by waveform80