waveform80/picamera #346: Can't capture long exposure at full resolution w/rPI camera module v2

c0de517e, 2016-11-10:
With the latest PiCamera and camera firmware the shutter speed is capped by the fps parameter one sets when creating the PiCamera object. While I can successfully create a PiCamera object at low fps when doing half-resolution captures, e.g.

PiCamera( resolution = '1640x1232' ,framerate=Fraction(0,1) )

the same doesn't work at full resolution, in fact, at full resolution ( resolution = '3280x2464' ) I don't seem to be able to use any framerate other than the default one (30fps)

nncanepa, 2016-11-13:
Look at [issue](https://github.com/waveform80/picamera/issues/336)

You need to increase gpu_mem in /boot/config.txt

waveform80, 2016-11-14:
Were you also getting an "out of resources" error on this? If so, I should add some more bits to the [Hardware limits](http://picamera.readthedocs.io/en/release-1.12/fov.html#hardware-limits) section which the [FAQ](http://picamera.readthedocs.io/en/release-1.12/faq.html) links to. Otherwise, this might warrant expanding that FAQ's description a bit.

c0de517e, 2016-11-15:
It worked indeed with 256mb of GPU memory! Note that I could capture full-res 30fps no problem w/128, it's just slow speeds that were affected. But still, it does work, I'll close this. Might I suggest for the future maybe printing a suggestion "try to increase gpu memory" when the exception is raised?

closed 2016-11-15 by c0de517e

waveform80, 2016-11-15:
Unfortunately that particular error code crops up in plenty of places where just raising the GPU memory won't make any difference (e.g. multiple processes accessing the camera). However, I will remove the "other than memory" bit from the error message because that's mis-leading. Sounds like I need to expand the notes in the Hardware limits section too...