waveform80/picamera #347: is there a way to mask an area for motion vector detection

cr0wm4n, 2016-11-13:
Hi,

Is there anyway to mask a portion of a frame for motion capture.

I am using the MyMotionDetector routine to detect motion

In the scene i am monitoring there is a highly light reflective area and this triggers motion detection.
If I try to use the sensitivity to stop motion being triggered then the rest of the frame is not sensitive enough.

Regards
Mike

waveform80, 2016-11-14:
Interesting question; I don't know if setting the [`zoom`](http://picamera.readthedocs.io/en/release-1.12/api_camera.html#picamera.PiCamera.zoom) parameter (which affects the region-of-interest or ROI of the sensor) also limits the area over which motion estimation is calculated. Even if it does, though, I'm guessing you wouldn't want to limit the field of view of your output (which `zoom` would do as it's a camera-level setting).

The array (`a`) that's being worked upon in [the recipe](http://picamera.readthedocs.io/en/release-1.12/recipes2.html#custom-outputs) is just a regular numpy array, so it shouldn't be hard to ignore portions of it by simply slicing the array, e.g. instead of `a['y']` use `a[5:, 5:]['y']` to ignore the top left 5x5 macro-blocks of the data.

cr0wm4n, 2016-11-21:
Thankyou for this :), 
im only just getting my head around numpy arrays , i spent all weekend reading up on them lol, 
and this makes sense, i will get it working :)

maybe its something that can be added in a later release, customizeable masks :)

Thankyou again
Mike

waveform80, 2016-11-21:
I don't think there's much point adding masks at the picamera level given it's simple to perform using numpy's slicing and selection operations. Anything I make in picamera is pretty much doomed to be much slower and less elegant than what you can manage directly in numpy. The alternative is for such features to be implemented in the firmware, which I could then expose via picamera, but that's not something I have any control over.

Just to demonstrate how easy masking is in numpy here's a demonstration of a couple of options. You can zero out data in a portion of an array (remember that arrays are indexed y,x instead of x,y, i.e. rows then columns):

```python
>>> import numpy as np
>>> a = np.arange(100).reshape((10, 10))
>>> a
    array([[ 0,  1,  2,  3,  4,  5,  6,  7,  8,  9],
       [10, 11, 12, 13, 14, 15, 16, 17, 18, 19],
       [20, 21, 22, 23, 24, 25, 26, 27, 28, 29],
       [30, 31, 32, 33, 34, 35, 36, 37, 38, 39],
       [40, 41, 42, 43, 44, 45, 46, 47, 48, 49],
       [50, 51, 52, 53, 54, 55, 56, 57, 58, 59],
       [60, 61, 62, 63, 64, 65, 66, 67, 68, 69],
       [70, 71, 72, 73, 74, 75, 76, 77, 78, 79],
       [80, 81, 82, 83, 84, 85, 86, 87, 88, 89],
       [90, 91, 92, 93, 94, 95, 96, 97, 98, 99]])
>>> a[2:5, 0:5] = 0
>>> a
    array([[ 0,  1,  2,  3,  4,  5,  6,  7,  8,  9],
       [10, 11, 12, 13, 14, 15, 16, 17, 18, 19],
       [ 0,  0,  0,  0,  0, 25, 26, 27, 28, 29],
       [ 0,  0,  0,  0,  0, 35, 36, 37, 38, 39],
       [ 0,  0,  0,  0,  0, 45, 46, 47, 48, 49],
       [50, 51, 52, 53, 54, 55, 56, 57, 58, 59],
       [60, 61, 62, 63, 64, 65, 66, 67, 68, 69],
       [70, 71, 72, 73, 74, 75, 76, 77, 78, 79],
       [80, 81, 82, 83, 84, 85, 86, 87, 88, 89],
       [90, 91, 92, 93, 94, 95, 96, 97, 98, 99]])
```

Alternatively, if you want to literally ignore those blocks for certain operations (e.g. taking a mean where simply counting the zeroed blocks would affect the result), you can construct a numpy mask array (an array with the same shape, but composed of bools) then use it for masking operations:

```python
>>> mask = np.zeros((10, 10), dtype=np.bool)
>>> mask[:, :] = True
>>> mask
    array([[ True,  True,  True,  True,  True,  True,  True,  True,  True, True],
       [ True,  True,  True,  True,  True,  True,  True,  True,  True, True],
       [ True,  True,  True,  True,  True,  True,  True,  True,  True, True],
       [ True,  True,  True,  True,  True,  True,  True,  True,  True, True],
       [ True,  True,  True,  True,  True,  True,  True,  True,  True, True],
       [ True,  True,  True,  True,  True,  True,  True,  True,  True, True],
       [ True,  True,  True,  True,  True,  True,  True,  True,  True, True],
       [ True,  True,  True,  True,  True,  True,  True,  True,  True, True],
       [ True,  True,  True,  True,  True,  True,  True,  True,  True, True],
       [ True,  True,  True,  True,  True,  True,  True,  True,  True, True]], dtype=bool)
>>> mask[2:5, 0:5] = False
>>> mask
    array([[ True,  True,  True,  True,  True,  True,  True,  True,  True, True],
       [ True,  True,  True,  True,  True,  True,  True,  True,  True, True],
       [False, False, False, False, False,  True,  True,  True,  True, True],
       [False, False, False, False, False,  True,  True,  True,  True, True],
       [False, False, False, False, False,  True,  True,  True,  True, True],
       [ True,  True,  True,  True,  True,  True,  True,  True,  True, True],
       [ True,  True,  True,  True,  True,  True,  True,  True,  True, True],
       [ True,  True,  True,  True,  True,  True,  True,  True,  True, True],
       [ True,  True,  True,  True,  True,  True,  True,  True,  True, True],
       [ True,  True,  True,  True,  True,  True,  True,  True,  True, True]], dtype=bool)
>>> a.mean()
    44.700000000000003
>>> a[mask].mean()
    52.588235294117645
```

In case that last example isn't clear - `a[mask]` means "all elements of the array `a` where the corresponding element in `mask` is `True`". Hence `a.mean()` is taking the mean of all elements of `a` including the ones we zeroed out, whilst `a[mask].mean()` is taking the mean of all elements of `a` ignoring entirely the ones we zeroed out.

I don't think I can do anything in picamera that would be as elegant as that, or for that matter as fast (numpy is compiled C, and has gone through many years of optimization efforts).

waveform80, 2016-11-21:
Oh, incidentally you can construct mask arrays really easily via comparison operators. So following on from the example above, you could construct another mask array from "all elements that aren't zero":

```python
>>> mask = a > 0
>>> mask
    array([[False,  True,  True,  True,  True,  True,  True,  True,  True, True],
       [ True,  True,  True,  True,  True,  True,  True,  True,  True, True],
       [False, False, False, False, False,  True,  True,  True,  True, True],
       [False, False, False, False, False,  True,  True,  True,  True, True],
       [False, False, False, False, False,  True,  True,  True,  True, True],
       [ True,  True,  True,  True,  True,  True,  True,  True,  True, True],
       [ True,  True,  True,  True,  True,  True,  True,  True,  True, True],
       [ True,  True,  True,  True,  True,  True,  True,  True,  True, True],
       [ True,  True,  True,  True,  True,  True,  True,  True,  True, True],
       [ True,  True,  True,  True,  True,  True,  True,  True,  True, True]], dtype=bool)
>>> a[mask].mean()
    53.214285714285715
```

cr0wm4n, 2016-11-21:
Thankyou so much :)

I havent been able to find anything as clear as these examples,
one more question, on a pi 3 would you expect using mean values as in your last example add much processing time and have much of a negative impact ?

cr0wm4n, 2016-11-21:
Sorted !!!

Ok i have a motion capture window resolution of 1296x730
this gives me 82x46 macro blocks

using your last example i initialised my mask like this 

   mask = np.zeros((motion_rows, motion_cols), dtype=np.bool)
   mask[:, :] = True
   mask[12:28, 0:26] = False  

then in MyMotionDetector i did this

      a = np.sqrt(
          np.square(a['x'].astype(np.float)) +
          np.square(a['y'].astype(np.float))
          ).clip(0, 255).astype(np.uint8)

      mean_mask = a[mask].mean()
      a[12:28, 0:26] = mean_mask

and now it ignores the reflections but is still sensitive everywhere i needed it :)

Thankyou again
Mike

waveform80, 2016-11-21:
numpy is *fast* (not as fast as the firmware's GPU processing, but about as fast as you'd hope to get on the ARM side of things). If you want to find out just how fast an operation is, I'd strongly recommend installing ipython (or ipython3) which has a very nice little `%timeit` function. The following is run on a Pi3 and times how long it would take to calculate the mean of the x and y vectors from the motion data associated with a 1280x720 recording:

```python
In [1]: import numpy as np

In [2]: from picamera.array import motion_dtype
In [3]: width = (1280 + 15) // 16
In [4]: height = (720 + 15) // 16

In [5]: vectors = np.array([(1, 2, 0) for row in range(height) for col in range(width)], dtype=motion_dtype).reshape((height, width))

In [6]: vectors
Out[6]: 
array([[(1, 2, 0), (1, 2, 0), (1, 2, 0), ..., (1, 2, 0), (1, 2, 0),
        (1, 2, 0)],
       [(1, 2, 0), (1, 2, 0), (1, 2, 0), ..., (1, 2, 0), (1, 2, 0),
        (1, 2, 0)],
       [(1, 2, 0), (1, 2, 0), (1, 2, 0), ..., (1, 2, 0), (1, 2, 0),
        (1, 2, 0)],
       ..., 
       [(1, 2, 0), (1, 2, 0), (1, 2, 0), ..., (1, 2, 0), (1, 2, 0),
        (1, 2, 0)],
       [(1, 2, 0), (1, 2, 0), (1, 2, 0), ..., (1, 2, 0), (1, 2, 0),
        (1, 2, 0)],
       [(1, 2, 0), (1, 2, 0), (1, 2, 0), ..., (1, 2, 0), (1, 2, 0),
        (1, 2, 0)]], 
      dtype=[('x', 'i1'), ('y', 'i1'), ('sad', '<u2')])

In [7]: %timeit vectors['x'].mean()
10000 loops, best of 3: 124 µs per loop

In [8]: %timeit vectors['y'].mean()
10000 loops, best of 3: 124 µs per loop

In [9]: mask = np.zeros((height, width), dtype=np.bool)

In [10]: mask[0:40, 5:35] = True

In [11]: %timeit vectors[mask]['x'].mean()
10000 loops, best of 3: 143 µs per loop
```

So there you go: calculating the mean of all x vectors or all y vectors from all the macro blocks of a 1280x720 recording takes about 120 µs. Calculating the mean of a masked portion of the array takes only slightly longer at approximately 140 µs. That's a small enough portion of a typical frame cycle to do real-time stuff.

However, be aware the picamera will probably be eating much more time than that just to get the data into the numpy array in the first place (transferring data from the MMAL layer to Python is one of the major costs). In other words: don't sweat too much about numpy's speed - it's *fast*. Worry more about how exactly you're using it (numpy has *many* ways of making things fast), and whether picamera (being relatively slow pure Python) has left you enough time for your processing.

cr0wm4n, 2016-11-21:
I will give ipython a go :)

and maybe i can use numpy as a buffer for capturing some audio samples now, I did try a while back when i was looking at pyaudio but got myself in a mess with it. ill give it ago again now its a bit clearer.

Thankyou for all your help.

Mike
,